antipole2/JavaScript_pi #3: Not working on Ubuntu 20.04

tai-fun, 2020-12-11:
Running O 5.2.4 I get the following runtime error

> 14:02:52.710 MESSAGE pluginmanager.cpp:1174 Checking plugin candidate: /usr/lib/opencpn/libJavaScript_pi.so
14:02:52.710 MESSAGE pluginmanager.cpp:2124 Plugin is compatible by elf library scan: true
14:02:52.710 MESSAGE pluginmanager.cpp:2232 PlugInManager: Loading PlugIn: /usr/lib/opencpn/libJavaScript_pi.so
14:02:52.710   ERROR dlunix.cpp:162 libwx_gtk2u_core-3.0.so.0: cannot open shared object file: No such file or directory
>
Ubuntu 20.04 only uses Gtk 3.

tai-fun, 2020-12-11:
The above was tested with the precompiled deb package.
Testing with compiling from source..... All is well.
So this just boils down to clearly name the packages, showing if they are compiled for gtk2 or gtk3.

antipole2, 2020-12-11:
Sorry about that muddle tai-fun.  I am not familiar with Ubuntu.  I have now included gtk2 in the installer name.
If you wish to contribute the gtk3 build, I will include it -  DM me on the forum.

tai-fun, 2020-12-12:
I'm attaching the Ubuntu 20.04 Gtk3 build here.
Gtk3 is slowly taking over in  all Linux distros.
With 20.04 Ubuntu stopped supporting gtk2, hence the transition problem,

antipole2, 2020-12-12:
Thanks - but cannot see the attachment.

antipole2, 2020-12-12:
Got the attachment and it is now added to the release.
Thank you

closed 2020-12-12 by tai-fun

antipole2, 2020-12-19:
The JavaScript plugin has been updated to v0.3.1 - details in a READ ME on the release page <https://github.com/antipole2/JavaScript_pi/releases/tag/v0.3>.

If you choose to rebuild and let me have a copy, I will update the build on the release page.

Regards, Tony

tai-fun, 2020-12-19:
Tony
It looks like you missed bumping the version in CMakeLists.txt.
It still reads
SET(VERSION_MAJOR "0")
SET(VERSION_MINOR "3.0")

End result is that the new version still has the same name as the previous
version.

BR/Thomas