antipole2/JavaScript_pi #16: Builds do NOT RUN due to permissions for upload

Rasbats, 2022-01-15:
Mike R. can build from his fork of the repo but Tony's builds do not start on CircleCI and possibly Appveyor.

[https://app.circleci.com/pipelines/github/antipole2/JavaScript_pi/13/workflows/6e120620-f5d6-4c32-82b0-e8df00a47376](https://app.circleci.com/pipelines/github/antipole2/JavaScript_pi/13/workflows/6e120620-f5d6-4c32-82b0-e8df00a47376)

leamas, 2022-01-15:
Mike R  (MR) has a cloudsmith user and an associated API key. The problem is that this user hasn't permissions to write at corresponding cloudsmith repositories like opencpn/Alpha etc.

There are two solutions. One is to talk to Rick so he gives MR the proper permissions. 

The other is the one I, Dave, Kees and perhaps others use: create your own Cloudsmith repositories  and use those instead. This is actually simple (and described in the manual).

Take whatever solution which works...

Rasbats, 2022-01-16:
@leamas : Thanks.

closed 2022-01-16 by Rasbats