antipole2/JavaScript_pi #53: Console windows do not stay on top of main frame

antipole2, 2023-02-26:


Rasbats, 2023-02-27:
`m_Console( wxWindow* parent, wxWindowID id = wxID_ANY, const wxString& title = wxT("JavaScript"), const wxPoint& pos = wxDefaultPosition, const wxSize& size = wxSize( 844,2000 ), long style = wxCAPTION|wxCLOSE_BOX|wxDEFAULT_DIALOG_STYLE|wxMINIMIZE_BOX|wxRESIZE_BORDER|wxSTAY_ON_TOP);`

Added in _JavaScriptgui.h_. Not yet tested.

antipole2, 2023-02-27:
Added through wxFormBuilder, which is more permanent than editing _JavaScriptgui.h_ but not sufficient.
Have asked in Zulip #Master - 5.8.0

antipole2, 2023-02-27:
Under OCPN5.6.2/wx3.1.5 console windows move with the frame.
Under OCPN5.7/x3.2 the consoles are not attached to the frame - at lease under MacOS.

antipole2, 2023-03-04:
Proper behaviour under wx3.2.
Changes made to v2.0.2 to accommodate this.

closed 2023-03-04 by antipole2