antipole2/JavaScript_pi #58: allow to push AIS NMEA messages

pterocarya-fraxinifolia, 2023-03-10:
the recent change
_OCPNpushNMEA( ) now checks the sentence starts "$.....,"_
inhibits to push AIS messages as they start with "!" rather than "$"

antipole2, 2023-03-10:
Oh dear!  Will make "!" legitimate alternative and issue update.
@pterocarya-fraxinifolia are you still using OCPN5.6.2 or have you moved on to the OCPN5.7+ betas?

pterocarya-fraxinifolia, 2023-03-10:
... I am still on 5.6.2

antipole2, 2023-03-14:
Update in:
*  v2.0.4 for OCPN5.6.2
*  v2.1.1+ for OCPN5.7.1+
For OCPN5.6.2 I failed to delete v2.0.1 from the catalogue.  Just choose the latest.

@pterocarya-fraxinifolia if you are happy with this, we can close this issue.

pterocarya-fraxinifolia, 2023-03-15:
Tony,
thanks a lot: the issue can be closed, I verified with v2.0.4 on 5.6.2
Bernd

antipole2, 2023-03-15:
Completed and verified

closed 2023-03-15 by antipole2