antipole2/JavaScript_pi #59: add parameter start, length to readtextfile()

pterocarya-fraxinifolia, 2023-03-10:
the readtextfile() seems to be limited to a size of 64k.
If this is the case, if would be helpful to add (optional?) parameters _start_ , _length_
in order to read chunks of a file larger than 64k

antipole2, 2023-03-10:
Looks like this should be possible but not a five minute job.  I _must_ attend to cruise planning, so may be some delay in implementing.
Temporary work-around: chop file up into instalments and read each.

pterocarya-fraxinifolia, 2023-03-10:
this is exactly what I did :-)
so: no need to hurry with the implementation

antipole2, 2023-10-10:
@pterocarya-fraxinifolia Can you explain what sort of data exceeding 64k you are processing?  And what is the symptom when you exceed 64k?
I could implement optional start and length parameters.
Am also considering a line-based approach: `readTextFile(file [, startLine, numberOfLines])` but not sure if that is more useful.

antipole2, 2024-01-03:
@pterocarya-fraxinifolia Please install the latest alpha build of the plugin from [my alpha repository](https://cloudsmith.io/~opencpn/repos/javascript-alpha/packages/).

You can then handle large files this way.  Let me know how it works for you.
[image]

pterocarya-fraxinifolia, 2024-01-04:
Hi Tony,
thanks a lot for the implementation: it looks great and does the job for me.
Bernd

antipole2, 2024-01-04:
Can you confirm if you are using my `.getTextLine( )`?  [You might have used` .getText(start, count)` and retained your own block management.]

pterocarya-fraxinifolia, 2024-01-04:
I indeed used .getText() initially, but .getTextLine() works as well on my
Windows 10 environment.
I will retest both on Debian in the next days.

closed 2024-04-03 by antipole2