antipole2/JavaScript_pi #60: Preferences - High Res Screen at 250%

rgleason, 2023-03-10:
Tony,  I've opened javascript and gone to prefs   on a high res screen  at 250% it would be good to have prefs resizeable so we can see the entire thing presented.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

antipole2, 2023-03-11:
Rick... "high res screen at 250%"
Not sure what this means.  How do I reproduce?
Tony

rgleason, 2023-03-11:
Tony, it will be hard to reproduce without a high res windows computer. I will try to find some code examples that fixed it. One thing that would help would be allowing the Preferences window user to resize in width and height.  Then we could see what else needed improvement.
[image]

rgleason, 2023-03-11:
Tony, I don't find a good example for you.

antipole2, 2023-03-12:
@rgleason ... not sure why the preferences dialogue shows in the odd sizes you see.  It looks like it is failing to adjust for the screen res.
But, as you suggested, I have made the dialogue resizable so you can drag to suit.  Each time you select a new panel within the dialogue, it will go back to what it thinks I the right size but at least you can view the dialogue.
Find the build in the [Alpha repository](https://cloudsmith.io/~opencpn/repos/javascript-alpha/packages/) - builds ending ce8a59b
Let me know and also if you have any thoughts on accounting for the screen res.

antipole2, 2023-03-15:
@rgleason Any progress on this?  Hope this is not bugging you :-)

rgleason, 2023-03-15:
That change has made it easier to understand how the plugin works.  I do have to resize the Prefs menu each time I hit a tab, I am not sure if that is your intent, but at least I can hit a tab and resize the window to fit so that I can read it.  I could show some screenshots of this if you would like, but I think you will understand what I mean.

In starting to get familiar with JS itself, I notice that the top menu, right side Icons seem to be very attenuated and small, probably because of the high res display. The two that are either side of the what I think are the "Tools" section are quite small.

[image]

I wish I knew some sample code to link to, to show how bdbcat handles Win10&11 scaling for icons and text fonts, which might help with these pesky items.  We had a Issue/Discussion on Opencpn about this here, but Dave probably knows the best way to handle this.  

This comes the closest to the point. There are other examples in OpenCPN issues which got resolved, 
DPI Aware OpenCPN. https://github.com/OpenCPN/OpenCPN/issues/2943

also I found some information about DPI Scaling in wxWidgets but I don't know if that is the way Dave solved it.
for some background  see this Issue   https://github.com/OpenCPN/OpenCPN/issues/2926 and down at the bottom the part about wxWidgets  which I copy to here:

> Also, for example Statusbar has some real text problems from a legibility standpoint. I think all of this is due to inconsistent use of wxWidgets DPI tools. Same thing with Ocpn_draw and other plugins. I don't think that I should have to go into every plugin and many OpenCPN text settings and change the font! It should use wxWidgets DPI tools to automatically scale the font up for my DPI! OpenCPN is not using the wxWidgets tools available.
> 
> wxWidgets has tools to handle DPI see https://github.com/jongough/ocpn_draw_pi/issues/513#issuecomment-1374692071
> and for Microsoft Windows https://github.com/jongough/ocpn_draw_pi/issues/513#issuecomment-1374692427
> 
> https://docs.wxwidgets.org/latest/overview_high_dpi.html
> Automatic scaling is convenient, but doesn't really allow the application to use the extra pixels available on the display. Visually, this means that the scaled application appears blurry, in contrast to sharper applications using the full display resolution, so a better solution for interpreting pixel values on high DPI displays is needed: one which allows to scale some pixel values (e.g. the total window size), but not some other ones (e.g. those used for drawing, which should remain unscaled to use the full available resolution).
> 
> Thus, the first thing do when preparing your application for high DPI support is to stop using raw pixel values, as they mean different things under different platforms when DPI scaling is used. Actually, using any pixel values is not recommended and replacing them with the values based on the text metrics, i.e. obtained using [wxWindow::GetTextExtent()](https://docs.wxwidgets.org/3.1.6/classwx_window.html#a4220668e42f3a173bf29d335da4db2c3), or expressing them in dialog units (see [wxWindow::ConvertDialogToPixels()](https://docs.wxwidgets.org/3.1.6/classwx_window.html#a3dbe034120d742c4f5f6d64dc5d69590)) is preferable
> 

Perhaps Dave @ bdbcat can advise and direct us to a reasonable example?

rgleason, 2023-03-15:
Also Jon's ocpn_draw had icon scalling problems too, but they were way to large on high res screens. He eventually resolved this problem satisfactorily and here was one of his observations  https://github.com/jongough/ocpn_draw_pi/issues/513#issuecomment-1381487228

I still don't know how he solved it, but it is working nicely now.  (Maybe we could find his pertinent changes in recent commits.?

rgleason, 2023-03-15:
PS: Tony, I wouldn't let these issues slow down release of your plugin.

antipole2, 2023-03-16:
Rick...  see this screenshot of the console bar.
[image]
The buttons to the right are just text buttons - not images.  So they should behave just like the ones to the left.  There is a 
spacer to separate the left block from the right.
This all works on MacOS and (according to Mike) on Windows and Linux.
Likely a wxWidgets issue?  Can you please let me have your OS and screen res so I can enquire.

Regarding the size of the tools window, I have found an unneccary setting width to 600 which is probably not a good idea.  Now removed.   Please give it a try [2.1.2+179.ffb7069](https://cloudsmith.io/~opencpn/repos/javascript-alpha/packages/?q=ffb7069)

rgleason, 2023-03-16:
Tony,  Screen res is 
[image]

rgleason, 2023-03-16:
Macs may be a bit different than this, but this is Windows..typically
1- minimize
2- maximize
3- close
[image]

rgleason, 2023-03-16:
Tony, good questions. I wont muddle the discussion.

antipole2, 2023-03-19:
I have had a go at taking screen resolution into account.  Builds (wrongly put into my Prod repository) [here](https://cloudsmith.io/~opencpn/repos/javascript-prod/packages/) - v2.1.3n.

This works fine on MacOS - but has nothing to do as MacOS takes care of it.  I am dependent on you @rgleason please for testing on high res monitor under Windows.
NB on first use, saved console position and size will be wrong until config saved - ignore

Please can you check the following:
1. After enabling the plugin, open Preferences/Tools
2. Tools should open in top right corner of screen and all be visible.  Width should be as shown here.
￼[image]- [ ]
3. On different tool tabs, the tools window should resize depth to accommodate the content but keeping the same width.  Help grows deep and Consoles will shrink back again.
4. Select the Diagnostics tab and then Dump.  This should display a dump window.
5. Scroll to the top, if needed.  Third line will show the scaling factor for the screen on which the tools window is displayed.
[image]
6. Use the Consoles tab to create a couple of new consoles.
7. They should be created in a sensible position and look like this . Note the three buttons right have been reworked.
[image]
8. Move and resize the consoles in such a way that they align with identifiable things on the canvas/frame,
9. Disable the plugin and re-eneable.  The consoles should be recreated in the same positions and with the same sizes.
10. Run the one line script  `alert(“Hello”);`  Move the alert box before dismissing it and run again.  The alert should appear in the same position from where it was dismissed.
11. Disable the plugin, reenable and run the script in the same console.  The alert should appear in the same place.
12. Park a couple of consoles using the Park button.  They should park starting near top left of the screen.  The location Is that chosen as default for Windows by Mike.
13. Use the Parking tab of the tools to choose your own parking place and spacing to suit your monitor
14. Disable and re-enable plugin.  Parking a console should use the same bespoke position
15. Locate aJavaScript .js file or use the Save button to save an existing script.
16. Now Load it back
17. Now load again.  This time it should offer the recents in a window like this:
[image]
End of basic screen resolution tests

rgleason, 2023-03-19:
Dear Tony. Wonderful effort. Thank you. I am away from my computer right now, but give me 2 days...thanks.

rgleason, 2023-03-20:
Tony,  I am back at the computer. Hope to try JS this afternoon, but I thought I would share some code to fix Dashboard that Hakan sent.  Some of this might provide some quidance if you find it useful.

DB: Preferences. Scale instr.list and +- icons #2983   [6a356104a63](https://github.com/OpenCPN/OpenCPN/commit/6a356104a637cfa45a9b1f75c583afea5e7df29c)

DB: Dialog "Add Instrument". Increase size to half Canvas hight #2983  [7fd1b159543](https://github.com/OpenCPN/OpenCPN/commit/7fd1b159543f3b22110e2d81e04b0330b6e24dd3)

DB: Preferences. Scale instr.list and +- icons https://github.com/OpenCPN/OpenCPN/issues/2983    [43a64cad1b3](https://github.com/OpenCPN/OpenCPN/commit/43a64cad1b3)

DB: Win Scaled fonts and sizing     [eb3b73acce2](https://github.com/OpenCPN/OpenCPN/commit/eb3b73acce2)

rgleason, 2023-03-20:
Tony, I tried download and install of .-  Prod repository) [here](https://cloudsmith.io/~opencpn/repos/javascript-prod/packages/) - v2.1.3n.  and I got errors. I think it is having trouble reading the initial icon perhaps.
[image]
[image]
[image]
[image]

Here is the log file.

[opencpn - Copy.log.txt](https://github.com/antipole2/JavaScript_pi/files/11022965/opencpn.-.Copy.log.txt)

antipole2, 2023-03-21:
Rick... I have downloaded the Mac tarball from the same build and all is OK so looks like a Windows issue.  Your log shows problems at 15:43 but all was well at 16:20, 16:24 & 17:54.  Maybe the first was an install issue but it has not stopped plugin activation at other times.
Your screenshots show the toolbar visible while the plugin manager window is open.  On Mac, the toolbar is never visible when the settings panel is open, so I am not sure at what point the plugin icon in the toolbar appears after activation.
But it looks like there is a problem with installing/setting up the toolbar icon.  I have not touched anything in that area since the previous build.
Things to check:
1. You still have access to the JS plugin Settings from the plugin manager.  You should be able to do those tests without the toolbar icon.  Also the dump (line 7?) shows whether it is using SVG or not.  Please advise.
2. Let's look at the support files - not sure where they are on Windows but in Mac they look like this
[image]

antipole2, 2023-03-21:
Rick...  I have seen no icon in the toolbar too - sometimes.  For me it seems to be related to using a recent build of OCPN +77a4b15.  I went back to Beta 4 and got the icons back.  I then went forward again to OCPN +77a4b15 and the icon was there.  Very strange.
Which version of OCPN are you using when you have the 'no icon' issue?  Please try an earlier one like Beta 4.

rgleason, 2023-03-21:
The JS icon is not loading for some reason.  It does exist in the "read-only" data directory.
C:\Users\fcgle\AppData\Local\opencpn\plugins\JavaScript_pi\data
I do not find a javascript_pi  read-write directory under 
C:\ProgramData\opencpn\plugins
but not all plugins require that.

Also the "Preferences" button does not seem to be working.  There seems to be some missing library.

```
10:31:13.023 MESSAGE plugin_loader.cpp:238 Loading plugins from dir: C:\Users\fcgle\AppData\Local\opencpn\plugins
10:31:13.023 MESSAGE plugin_loader.cpp:405 PlugInManager searching for PlugIns in location C:\Users\fcgle\AppData\Local\opencpn\plugins
10:31:13.026 MESSAGE plugin_loader.cpp:443 Found 10 candidates
10:31:13.026 MESSAGE plugin_loader.cpp:258 Checking plugin candidate: C:\Users\fcgle\AppData\Local\opencpn\plugins\JavaScript_pi.dll
10:31:13.026 MESSAGE plugin_loader.cpp:1157 Plugin is compatible: true
10:31:13.026 MESSAGE plugin_loader.cpp:1172 PlugInManager: Loading PlugIn: C:\Users\fcgle\AppData\Local\opencpn\plugins\JavaScript_pi.dll
_**10:31:13.026   ERROR dlmsw.cpp:167 Failed to load shared library 'C:\Users\fcgle\AppData\Local\opencpn\plugins\JavaScript_pi.dll' (error 127: The specified procedure could not be found.)**_
10:31:13.026 MESSAGE plugin_loader.cpp:1226    PlugInManager: Cannot load library: C:\Users\fcgle\AppData\Local\opencpn\plugins\JavaScript_pi.dll
10:31:13.026 MESSAGE plugin_loader.cpp:258 Checking plugin candidate: C:\Users\fcgle\AppData\Local\opencpn\plugins\otcurrent_pi.dll
```

rgleason, 2023-03-21:
If you would like I could clone and try to build javascript with a debug pdb and then run it. Might be tomorrow as we are traveling this afternoon.

antipole2, 2023-03-21:
Very kind of you Rick.  To date, Mike has done the Windows debugging for me, so I have invited him to look at where this is at.  He might want to keep on top of it or may be glad if you look.  Expect him to chip in here soon.

rgleason, 2023-03-21:
Tony, Doesn't this help?

```
There seems to be some missing library.
10:31:13.026 ERROR dlmsw.cpp:167 Failed to load shared library 'C:\Users\fcgle\AppData\Local\opencpn\plugins\JavaScript_pi.dll' (error 127: The specified procedure could not be found.)
```

rgleason, 2023-03-21:
Plugin_loader.cpp  Line 375
#ifndef CLIAPP
      // The CLI has no graphics context, but plugins assumes there is.
      if (pic->m_bEnabled) {
        pic->m_cap_flag = pic->m_pplugin->Init();     <----- HERE
        pic->m_bInitState = true;
        evt_load_plugin.Notify(pic);
      }
#endif
[image]

rgleason, 2023-03-21:
I went to beta 4, and reinstalled and that is what I used.  I also cleaned out the plugins.

antipole2, 2023-03-21:
OK - let's wait for Mike's input.

rgleason, 2023-03-21:
OK,  but in windows we don't have libs in our plugins folder.
`That plugins folder should contain the libraries for all installed plugins. Seems like the plugin installed has failed to put the library in there. Are the others there?`

antipole2, 2023-03-21:
Are the libraries for other plugins in there?

Rasbats, 2023-03-21:
Which branch and which commit is it best to play with?

antipole2, 2023-03-21:
Branch v2.1 latest version v2.1.3n (ff09a1a I think).  Latest remote build in my prod repository (prod was an accident).  Changes since it all worked relate only to hi res display and should not have touched things like plugin loading ?-)

Rasbats, 2023-03-21:
Thanks. Just made a tarball for testing.

Rasbats, 2023-03-21:
Crash! Reverted the changes for hi res display and everything works here. Now to find what has changed.

antipole2, 2023-03-21:
I'm actually in the middle of reworking the hi res stuff.
It turns out that although ToDIP and FromDIP work OK for me and it knows MacOS takes care of it,
the ->GetDPIScaleFactor() does not. That is, it returns 2 on my high res display.
Am having to use a platform-dependent macro for it.

So if you can tell me what was causing the problem, that's great.  But I am unlikely to be able to use your code changes..

antipole2, 2023-03-21:
Mike... I have finished reworking the scaling stuff.  If you are able to identify what caused the issue in my original approach, I will check what I might need to change in the way I do it now,
I am finishing for this evening and will not be available tomorrow morning.

Rasbats, 2023-03-23:
For the constructor in _JavaScriptgui_impl.h_ (about line 254) I just need to remove the FromDIP for consolePosition, consoleSize.

```
    Console(wxWindow *parent, wxString consoleName,
    		wxPoint consolePosition = pJavaScript_pi->m_parent_window->ToDIP(wxPoint(300,100)),
    		wxSize consoleSize = pJavaScript_pi->m_parent_window->ToDIP(wxSize(738,800)),
    		wxPoint dialogPosition = wxPoint(150, 100), wxPoint alertPosition = wxPoint(90, 20), wxString fileString = wxEmptyString,
    		bool autoRun = false, wxString welcome = wxEmptyString, bool parked = false)
    		: m_Console(parent, wxID_ANY, consoleName, consolePosition, consoleSize,
    		wxCAPTION|wxRESIZE_BORDER|wxCLOSE_BOX|wxMINIMIZE|wxSYSTEM_MENU)
```
Plugin runs fine here. consoleSize is saved correctly but not consolePosition.

antipole2, 2023-03-25:
Mike... the issue was that in the contstructor I was using FromDIP to convert my default position & size to pass to the base constructor.  But FromDIP is a method of the window being constructed but not yet existing.  I got away with it on MacOS as it does nothing as MacOS takes care of the rescaling.  Not so on Windows and probably Linux.

My Mk 2 approach was to do the conversion based on the parent window before calling the constructor.  But that meant I had to not use defaults and lots or code required in the several places from where I construct consoles.

Having done all that, I realised that I should be able to do the conversion on the constructor if I use the parent window that already exists.  So now kind of back to Mk 3.  At the same time I moved to use #defines for the various positions and sizes.  So now I have:
````
    Console(wxWindow *parent, wxString consoleName,
    		wxPoint consolePosition = pJavaScript_pi->m_parent_window->FromDIP(NEW_CONSOLE_POSITION),
    		wxSize consoleSize = pJavaScript_pi->m_parent_window->FromDIP(NEW_CONSOLE_SIZE),
    		wxPoint dialogPosition = pJavaScript_pi->m_parent_window->FromDIP(DEFAULT_DIALOG_POSITION),
    		wxPoint alertPosition = pJavaScript_pi->m_parent_window->FromDIP(DEFAULT_ALERT_POSITION),
    		wxString fileString = wxEmptyString, bool autoRun = false, wxString welcome = wxEmptyString, bool parked = false)
    		: m_Console(parent, wxID_ANY, consoleName, consolePosition, consoleSize,
    		wxCAPTION|wxRESIZE_BORDER|wxCLOSE_BOX|wxMINIMIZE|wxSYSTEM_MENU)
        {
````
Build [a3843f1](https://cloudsmith.io/~opencpn/repos/javascript-alpha/packages/?q=version%3A2.1.3%2B2877.a3843f1) in my alpha repository from branch v2.1.
Mike... can you please check it out on Windows with standard screen before I ask Rick to try on his Hi res Windows?
A check on Linux would also be wise.

rgleason, 2023-03-25:
Watching. Thank you Tony. Will give it a spin.

Rasbats, 2023-03-26:
No problem with icons here. Perhaps Rick needs to restart OpenCPN.
JS works fine on the low res screen. Retains size of the initial dialog but the position is not saved when OpenCPN is restarted.

antipole2, 2023-03-26:
>  Retains size of the initial dialog but the position is not saved when OpenCPN is restarted.
Working fine for me.  Difficult to see how this could be a Windows issue.  Please recheck and say exact circumstances.
NB Position is now relative to the screen, so, if the frame/canvas has moved after restart, the position relative to that will be different.

Rasbats, 2023-03-26:
The console position is recorded after the change of position but is not used to position the console when OpenCPN is restarted.
```
JavaScript\:ConsolePosX=68
JavaScript\:ConsolePosY=67
JavaScript\:ConsoleSizeX=792
JavaScript\:ConsoleSizeY=713
JavaScript\:DialogPosX=150
JavaScript\:DialogPosY=100

JavaScript\:ConsolePosX=954
JavaScript\:ConsolePosY=156
JavaScript\:ConsoleSizeX=808
JavaScript\:ConsoleSizeY=753
JavaScript\:DialogPosX=150
JavaScript\:DialogPosY=100

```
Start the program with console at upper left. Move the console. Restart OpenCPN and it is back where it started.

[image]
[image]

antipole2, 2023-03-26:
I have realised that the FromDIP correction was only being applied to the default parameters.  I have now moved it to  the base constructor call.  Build [290aad3](https://cloudsmith.io/~opencpn/repos/javascript-alpha/packages/?q=version%3A2.1.3%2B2911.290aad3).
But that does not explain what you Mike are seeing.  It suggests that the default parameter is not being replaced by the supplied parameter - a C++ issue.  Please check with the latest build and confirm whether the console size is being restored as well as its position.  If still a problem with Windows, I will have to do an explicit move post-construction.

Rasbats, 2023-03-26:
Same result as before:
SAVED AFTER OPENCPN CLOSED:
```
[PlugIns/JavaScript_pi]
VersionMajor=2
VersionMinor=1
ShowJavaScriptIcon=1
CurrentDirectory=D:\\Documents
ParkingBespoke=1
ParkingMinHeight=39
ParkingStub=63
ParkingLevel=35
ParkingFirstX=85
ParkingSep=2
JavaScript\:Parked=0
JavaScript\:ConsolePosX=1242
JavaScript\:ConsolePosY=398
JavaScript\:ConsoleSizeX=588
JavaScript\:ConsoleSizeY=325
JavaScript\:DialogPosX=150
JavaScript\:DialogPosY=100
JavaScript\:AlertPosX=90
JavaScript\:AlertPosY=20
JavaScript\:AutoRun=0
JavaScript\:LoadFile=
Consoles=JavaScript
```
WHEN RESTARTED:
```
[PlugIns/JavaScript_pi]
VersionMajor=2
VersionMinor=1
ShowJavaScriptIcon=1
CurrentDirectory=D:\\Documents
ParkingBespoke=1
ParkingMinHeight=39
ParkingStub=63
ParkingLevel=35
ParkingFirstX=85
ParkingSep=2
JavaScript\:Parked=0
JavaScript\:ConsolePosX=68
JavaScript\:ConsolePosY=67
JavaScript\:ConsoleSizeX=604
JavaScript\:ConsoleSizeY=365
JavaScript\:DialogPosX=150
JavaScript\:DialogPosY=100
JavaScript\:AlertPosX=90
JavaScript\:AlertPosY=20
JavaScript\:AutoRun=0
JavaScript\:LoadFile=
Consoles=JavaScript
```
If I just install the updated tarball without removing the .ini data I notice that the console starts minimised.

antipole2, 2023-03-26:
I suspect you could have some non-visual  corruption in the .ini file.  If it cannot read a parameter, it will fall back to the default.
Please quite OCPN, deleted the entire JavaScript_pi section and test again.

Rasbats, 2023-03-26:
The same error with position happens after the .ini is rebuilt.
`ScreenToFrame` is used in SaveConfig(), for the position but not the size. Is this causing an issue? It doesn't seem to have any use.

rgleason, 2023-03-26:
I tried the Windows version 5.7.1 with the javascript settings removed in the ini file, and the plugin acts the same as before, no icon shown in the Toolbar when enabled.

antipole2, 2023-03-26:
> I tried the Windows version 5.7.1 with the javascript settings removed in the ini file, and the plugin acts the same as before, no icon shown in the Toolbar when enabled.

Rick... I take it you mean no icon rather than a plain black one without the {JS} text, as I see on older versions of MacOS?
No icon at all - over to MIke on that one, please.  It feels Windows-ish.

I will look at building into the tools a way to do the same as clicking on the icon, so we can test even when the icon is missing.

antipole2, 2023-03-26:
> The same error with position happens after the .ini is rebuilt.
> `ScreenToFrame` is used in SaveConfig(), for the position but not the size. Is this causing an issue? It doesn't seem to have any use.

`ScreenToFrame' now does nothing and, anyway, only applied to positions, not sizes.
Branch v2.1-26-Mar-tests has tracing of console build calls from loadConfig.  Please can you do local build from that and report what is happening.  My trace is:
```
L33	LoadConfig Console PluginTests to be constructed at position 153 23 with size 153 23
L33	LoadConfig Console PluginTests was constructed at position 153 23 with size 153 23
L33	setConsoleMinSize text PluginTests, size X:69 Y: 16
L33	LoadConfig Console A1 to be constructed at position 99 23 with size 99 23
L33	LoadConfig Console A1 was constructed at position 99 23 with size 99 23
L33	setConsoleMinSize text A1, size X:15 Y: 16
```

antipole2, 2023-03-27:
> I tried the Windows version 5.7.1 with the javascript settings removed in the ini file, and the plugin acts the same as before, no icon shown in the Toolbar when enabled.

Rick... are you OK now to do testing as per my list of tests h[ere](https://github.com/antipole2/JavaScript_pi/issues/60#issuecomment-1475163895) but with latest build?
If you are still having problem with no entry in the toolbar, I have a work-around in the works for you.

rgleason, 2023-03-28:
Tony, I found that I had to remove the old javascript plugin that failed before the new one could be installed (as per your post in OpenCPN).   Now after clearing https://github.com/OpenCPN/OpenCPN/issues/3114#issuecomment-1485075893  and installing
and was finally able to install javascript-2.1.3+186.290aad3_msvc-wx32-10-win32.tar.gz
successfully and show a JS icon rather than a green puzzle!
This had been plaguing me for quite awhile. Rasbats could install the plugin successfully, but I could not.
It is now clear that I should have removed all previous instances of javascript from these locations.

It seems to be working nicely now. I would like to show how it is working now.
I think it is entirely useable, but could have some improvement. There are some funny characters in certain fields and some fields that need to be filled in seem to be somewhat short.

[image]

[image]

[image]

[image]

[image]

[image]

[image]

Below are Resized and Enlarged

[image]

[image]

[image]

[image]

[image]

antipole2, 2023-03-28:
Very helpful, Rick.  But it has opened up the issue that wxFormBuilder-generated forms do not take resolution into account.  To fix this up could be very tedious and messy - lots of poking into the wxForBuilder-generated code.  Ugh!  I will have to think about this before embarking on a solution.

Please can you now do the other tests from 4 onwards in [my list](https://github.com/antipole2/JavaScript_pi/issues/60#issuecomment-1475163895)?

I can foresee issues with JS dialogues (not yet in the test list) but I need the result from Step 5 before working on that.

@Rasbats The the 'funny characters' in the last two screenshots - I thought you and I had fixed that.  Can you please check whether you are seeing them these days?

Rasbats, 2023-03-28:
Trace with blank .ini file when starting OpenCPN:

```
L4	Constructed console JavaScript size x 738 y 600  minSize x -1 y 10
L4	setConsoleMinSize text JavaScript, size X:52 Y: 15
L4	Captured openCPNConfig
L4	JavaScript->isWaiting() doing full check
L4	JavaScript->isWaiting() returning new result false
L4	Captured openCPNConfig
L4	Captured openCPNConfig

```

JS starts minimized.

Reposition the first console, close and then restart:

```
L4	Loaded parking config ParkingBespoke:true ParkingMinHeight:39, ParkingStub:63 
L4	Constructed console JavaScript size x 815 y 590  minSize x -1 y 10
L4	setConsoleMinSize text JavaScript, size X:52 Y: 15
L4	Captured openCPNConfig
L4	JavaScript->isWaiting() doing full check
L4	JavaScript->isWaiting() returning new result false
L4	Captured openCPNConfig
L4	Captured openCPNConfig

```
No indications of position, unlike your trace.

I will reopen the issue of characters with a separate issue.

antipole2, 2023-03-28:
Rick...please[ see here](https://github.com/antipole2/JavaScript_pi/issues/64#issue-1643688235).

rgleason, 2023-03-28:
Morning Tony,
`Very helpful, Rick. But it has opened up the issue that wxFormBuilder-generated forms do not take resolution into account. To fix this up could be very tedious and messy - lots of poking into the wxForBuilder-generated code. Ugh! I will have to think about this before embarking on a solution.`

Understood.  Will do other tasks.

rgleason, 2023-03-28:
When I "Load" and pick Other and navigate to C:\Users\fcgle\AppData\Local\opencpn\plugins\JavaScript_pi\data\scripts

[image]

antipole2, 2023-03-28:
>  should see shouldn't I?
Navigate to where you have some scripts, such as your dialogue.js file and load it.

rgleason, 2023-03-28:
[image]
[JS-after-disable.txt](https://github.com/antipole2/JavaScript_pi/files/11090079/JS-after-disable.txt)

antipole2, 2023-03-28:
Rick: Your open dialogue seems to have opened on the built-in scripts inside the plugin installation - same location as the icons.  That is just how the open happened to remember where it had been last?  Those internal scripts are used by the `require` function and do not have a .js subscript.  Please see the User Guide sections 4 & 5.

Thank you very much for all the screenshots and dumps.  Lots for me to work on.  Will get back in due course.  Trouble is, I must get on with preparing for our summer cruise and once away from late April for four months will not be able to do more on this.

rgleason, 2023-03-28:
> Your open dialogue seems to have opened on the built-in scripts inside the plugin installation - same location as the icons.

Yes, I browsed to there.   Will read the manual.  I am envious 4 months cruise! Awesome. I still have to get the boat ready, last year I worked on it throughout March, not so this year.

rgleason, 2023-03-28:
10. Run the one line script  alert(“Hello”);  Move the alert box before dismissing it and run again. The alert should appear in the same position from where it was dismissed.
Ran one line multiple times. Then tried resize to be smaller, Then ran it again and the screen resizes to the standard size.  Perhaps this is expected?
[image]

[image]

However I am not finding the alert.  What is it? What does it look like?

antipole2, 2023-03-28:
Rick: you are running the initial print("Hello... script, which outputs to the output pane.  If the console is too small to see the output, it grows itself to make sure you can see the output.

The test is to run a different one-line script (replacing the default script)
`alert("Hello");`
It should display a yellow alert window.  It is that that whose position is being tested.

rgleason, 2023-03-28:
14. Park a couple of consoles using the Park button. They should park starting near top left of the screen. The location Is that chosen as default for Windows by Mike.
15. Use the Parking tab of the tools to choose your own parking place and spacing to suit your monitor

```
Bespoke settings

#define CONSOLE_MIN_HEIGHT 35L
#define CONSOLE_STUB 66L
#define PARK_LEVEL 44L
#define PARK_FIRST_X 36L
#define PARK_SEP 10L
The above are all in DIP pixels
```
[image]

[15 more replies]

closed 2023-04-01 by antipole2