antipole2/JavaScript_pi #63: Funny characters in some dialogs

Rasbats, 2023-03-28:
[https://github.com/antipole2/JavaScript_pi/issues/60](https://github.com/antipole2/JavaScript_pi/issues/60)

See screen shots from @rgleason.

Rasbats, 2023-03-28:
These quotes have not been 'translated' for Windows:
e.g. “Latitude” (It won't allow me to copy the ones used in the code! Perhaps just use single quotes.

antipole2, 2023-03-28:
They originate from with wxFormBuilder.  On close examination therein, it looks like they got in as 'smart quotes'.
I have replaced them with straight ones, I think.  We will; see in the next build.
I do not want to use single quotes as they illustrate a JSON message, which uses the quotation mark.

antipole2, 2023-03-31:
Should be fixed in  [alpha build bb9e490](https://cloudsmith.io/~opencpn/repos/javascript-alpha/packages/?page=1) for the message tab.
No fix possible for the Diagnostics string - displaying these characters on Windows IS the problem!

rgleason, 2023-03-31:
[image]

Tight fields for entry.  Requires resize to pull down to see last line.
[image]

Requires resize
[image]

Nmea  after a resize.  It is just a thin strip when first opened.
[image]

How Message Opens.
[image]

How parking opens
[image]

How Help opens.
[image]

How diagonstics opens.
[image]

Diagnostics resized.
[image]

Shrink  Minimize
[image]

Expand
[image]

Load Script
[image]

Favorites
[image]

Park expanded and resized.
[image]

Dump
[image]

Run script register, fields are too small for entry.
[image]

Message result
[image]

```
JavaScript plugin version 2.1
JavaScript patch 3
JavaScript tools window DPI scaling factor 2.500000
wxWidgets version 3.2.1.0 or 3201
OCPN API version 1.18
Duktape version 20700
Not using svg
pJavaScript_pi->m_pconfig			0x000134f290
pJavaScript_pi->m_parent_window		0x00039cfc40
favouriteFiles:
	C:\Users\fcgle\Downloads\dialogue_demo.js
	C:\Users\fcgle\Downloads\yet-another-demo.js
recentFiles:
	C:\Users\fcgle\Downloads\another-demo.js
	C:\Users\fcgle\Downloads\second-dialog-demo.js
pJavaScript_pi->mpFirstConsole	0x000a368818

----------Console three----------
Console name:		three
this:					0x000a368818
mpNextConsole:		0x0016102018
mStatus:				DONE 
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	1
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:36 y:44	DIP x:14 y:18
Size():				x:241 y:88	DIP	x:96 y:35	MinSize() x:241 y:87
m_parked:			true position x:14 y:18
isParked():			true
Messages callback table
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		true
m_result:				Registration was cancelled
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------Console seconname----------
Console name:		seconname
this:					0x0016102018
mpNextConsole:		nullptr
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	0
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:85 y:160	DIP x:34 y:64
Size():				x:1877 y:1588	DIP	x:751 y:635	MinSize() x:324 y:87
m_parked:			false position x:0 y:0
isParked():			false
Messages callback table
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------------------------------------------------

pJavaScript_pi->mpBin		nullptr

End of dump
```

rgleason, 2023-03-31:
Sorry used wrong issue.

antipole2, 2023-04-21:
Deleted comments not relevant to this issue.
Issue resolved in current version.

closed 2023-04-21 by antipole2