antipole2/JavaScript_pi #65: Idea: Create entry console into SignalK (for boat data and units) shared with other PI

rgleason, 2023-03-28:
1. Use a Javascript entry console to allow user entry (and changing) of essential boat data and unit preferences into a section of the signalK xml.
2. Save that section of the SignalK xml uniquely named, to a single location in the opencpn support files.
3. Provide access to this information by all plugins so enabled, via the API or some other means.

rgleason, 2023-03-28:
[FS#920 - UI UNITS Setting measurement units](https://opencpn.org/flyspray/index.php?do=details&task_id=920&string=units&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=)

Make the possibility of a global setting Units of measurement in OpenCPN. someone accustomed to use the metric system of measurement -

    Distance: kilometers and meters, feet, yards, nautical miles nautical miles and feet, nautical miles and yards.
    Spot soundings: feet, meters, fathoms.
    Mast Height: meters, feet.
    Speed​​: knots, kilometers per hour, miles per hour.
    Bearing: Magnetic, true
    Fuel: Gallons, Liters
    Position: degrees minutes seconds, degrees and minutes, degrees.
    Time: 12 hours, 24 hours
    Date Format: YYYY-MM-DD or mm-dd-yyyy
    Temperature: Fahrenheit, Celsius.
    Wind speed: knots, kilometers per hour, meters per second.
    Pressure: mBar, mmHg, Pa.
    Current Velocity: knots, kilometers per hour, miles per hour.

Also NOTE: Related but separate OwnShip Settings [FS#1102](https://opencpn.org/flyspray/index.php?do=details&task_id=1102&string=units&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=)

Consider this note requesting Small distance units [FS#2407](https://opencpn.org/flyspray/index.php?do=details&task_id=2407&string=units&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=)

[FS#1102 - UI Option Ship Tab Boat Data- Draft, Speed, Clearance, Displacement ](https://opencpn.org/flyspray/index.php?do=details&task_id=1102&string=units&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=)

Hello, On this tab, the user can already type the :

    Length Over All
    Width Over All for its own boat.

It would be nice to have the ability to add :

    minimum draft
    speed
    clearance
    displacement

Consolidated List  -Parameters adjustable should be:

    Name (from AIS, if available)
    Clearance (aerial)
    Draft (underwater)
    Beam
    Overall length LOA -done
    Waterline length
    Displacement
    Max Hull speed
    Average Hull speed
    Minimum Sailing speed (change to power)
    Keel type [Full,Fin,Bulb,Canting]
    GPS Position (from the bow and from centerline) -done
    Radar Position & Height (from the bow and from centerline)
    AIS Position & Height (from the bow and from centerline)
    AIS standard ship shapes
    AIS minimum size to show the 'custom shape' (to ensure a correct display of it) otherwise, display the standard shape.

Plugin API access to these values, so they can be used and edited.

NOTE: See related but separate UI UNITS Setting [FS#920](https://opencpn.org/flyspray/index.php?do=details&task_id=920&string=units&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=)

[ FS#2407 - UI Small distance units](https://opencpn.org/flyspray/index.php?do=details&task_id=2407&string=units&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=)

E.g. Nobeltec Admiral has own unit for small distances - less than 0.2 nm. It is easier to understand distance 250 m than 0.13 nm specially below 100 m distances. Possible choices should be the same as for distance, but I think feet should be included for US.

[ FS#515 - UI UNITS Allow routes, measurements to be in statute miles](https://opencpn.org/flyspray/index.php?do=details&task_id=515&string=units&type%5B0%5D=&sev%5B0%5D=&pri%5B0%5D=&due%5B0%5D=&reported%5B0%5D=&cat%5B0%5D=&status%5B0%5D=open&percent%5B0%5D=&opened=&dev=&closed=&duedatefrom=&duedateto=&changedfrom=&changedto=&openedfrom=&openedto=&closedfrom=&closedto=)
Allow routes and measurements to be in statute miles as well as nautical miles. This request is similar to the current request for kilometers. The Great Lakes charts are in statute miles and need to provide vessel traffic services with statute miles (and times derived from statute miles).

antipole2, 2023-03-28:
Rick... if you want to write .xml files, have a look at my TracksToKML utility in the [shared scripts library](https://github.com/antipole2/JavaScripts-shared/blob/main/library/library_index.adoc).

Here I am outputting tracks in KML.  The tracks part is not what you are wanting but it does write KML, which gives you a lead on writing XML.

antipole2, 2023-03-28:
Rick...  I don't think I will be putting effort in here, but you should be able to do what you want.  If you feel the plugin lacks something you need, raise it as an enhancement request in this issues list.

Note there is an example in the shared scripts library of saving preferences and loading them back.  You could use this to save your data input and return to it to update.  Or you could parse the XML output and repopulate the dialogue.

Note too that the plugin can send/receive OCPN messages to OCPN or another plugin.

antipole2, 2023-04-03:
Your script runs OK by me.
onDialogue does not support scrolling of the dialogue or two columns.  There is no such thing in wxDialog which it uses.
The scrollong you see in the JS consoles is within the text fields, not the whole widlow.

With a large dialogue, keeping track of the index in order to readback becomes tricky and brittle.  If you add a line, it wold screw up the indexes.  So I tend to constrauct the dialogue, noting the index at each step, viz:
```javascript
dl = [
	{type:"caption", value:"Selector"},
	]
boundaryIndex = dl.length;
dl.push({type:"radio", label:"Select boundary to use", value:boundaryNames});
inOutIndex = dl.length;
dl.push({type:"radio", label:"Inside or outside boundary", value:["Inside", "Outside"]});
crossIndex = dl.length;
dl.push({type:"tick", value:["Include routes crossing boundary"]});
````
Here I am noting the  index I will need to access the value of the item.

> What resource should I use for looking commands up.

Don't understand question.

rgleason, 2023-04-03:
Would this be a good reference?
https://javascript.info/

antipole2, 2024-04-23:
@rgleason Are you OK to close this?

closed 2024-04-23 by rgleason