antipole2/JavaScript_pi #67: Consoles not being restored to correct place on plugin activation on Windows Hi Res screen

antipole2, 2023-03-31:
Rick reports consoles all stacked against left side of screen on plugin activation.

antipole2, 2023-03-31:
@rgleason
Try [alpha build bb9e490](https://cloudsmith.io/~opencpn/repos/javascript-alpha/packages/?page=1)
Rick Sorry... need to ask for something else to be sure to nail the console positioning issues.
Please:

1. Have a couple of consoles somewhere (different positions, different sizes).
2. Park a third console using the P button.
3. Screenshot
4. Take a dump
5. Quit OPCN
6. Copy me the plugin's section of the .ini file
7. Restart OCPN and activate the plugin
8. Screenshot
9. Take a dump

That is all working fine on Mac and Windows standard display but it is likely I have not covered all issues with the DIP conversions. The above should let me check it all out.

rgleason, 2023-03-31:
[image]
[image]

JavaScript plugin version 2.1
JavaScript patch 3
JavaScript tools window DPI scaling factor 2.500000
wxWidgets version 3.2.1.0 or 3201
OCPN API version 1.18
Duktape version 20700
Not using svg
pJavaScript_pi->m_pconfig			0x00015fdb88
pJavaScript_pi->m_parent_window		0x0003bbd510
favouriteFiles:
	C:\Users\fcgle\Downloads\dialogue_demo.js
	C:\Users\fcgle\Downloads\yet-another-demo.js
recentFiles:
	C:\Users\fcgle\Downloads\another-demo.js
	C:\Users\fcgle\Downloads\second-dialog-demo.js
pJavaScript_pi->mpFirstConsole	0x000157ea00

----------Console three----------
Console name:		three
this:					0x000157ea00
mpNextConsole:		0x001adbb600
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	7602281
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:2078 y:1216	DIP x:831 y:486
Size():				x:1765 y:916	DIP	x:706 y:366	MinSize() x:241 y:87
m_parked:			false position x:0 y:0
isParked():			false
Messages callback table
	OCPN_CORE_SIGNALK		
	OCPN_OPENGL_CONFIG		
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------Console seconname----------
Console name:		seconname
this:					0x001adbb600
mpNextConsole:		0x001ade0550
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	1936028192
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:36 y:44	DIP x:14 y:18
Size():				x:324 y:88	DIP	x:130 y:35	MinSize() x:324 y:87
m_parked:			true position x:14 y:18
isParked():			true
Messages callback table
	OCPN_CORE_SIGNALK		
	OCPN_OPENGL_CONFIG		
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------Console another----------
Console name:		another
this:					0x001ade0550
mpNextConsole:		0x001adf5c48
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	0
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:-14 y:1492	DIP x:-6 y:597
Size():				x:1457 y:743	DIP	x:583 y:297	MinSize() x:275 y:87
m_parked:			false position x:0 y:0
isParked():			false
Messages callback table
	OCPN_CORE_SIGNALK		
	OCPN_OPENGL_CONFIG		
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------Console third----------
Console name:		third
this:					0x001adf5c48
mpNextConsole:		nullptr
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	538976288
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:370 y:44	DIP x:148 y:18
Size():				x:234 y:88	DIP	x:94 y:35	MinSize() x:234 y:87
m_parked:			true position x:148 y:18
isParked():			true
Messages callback table
	OCPN_CORE_SIGNALK		
	OCPN_OPENGL_CONFIG		
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------------------------------------------------

pJavaScript_pi->mpBin		nullptr

End of dump

rgleason, 2023-03-31:
[PlugIns/JavaScript_pi]
VersionMajor=2
VersionMinor=1
ShowJavaScriptIcon=1
CurrentDirectory=C:\\Users\\fcgle\\OneDrive\\Documents
ShowingConsoles=0
Recents=C:\\Users\\fcgle\\Downloads\\another-demo.js;C:\\Users\\fcgle\\Downloads\\second-dialog-demo.js
Favourites=C:\\Users\\fcgle\\Downloads\\dialogue_demo.js;C:\\Users\\fcgle\\Downloads\\yet-another-demo.js
ParkingBespoke=1
ParkingMinHeight=35
ParkingStub=66
ParkingLevel=44
ParkingFirstX=36
ParkingSep=10
three\:Parked=0
three\:ConsolePosX=831
three\:ConsolePosY=486
three\:ConsoleSizeX=706
three\:ConsoleSizeY=366
three\:DialogPosX=630
three\:DialogPosY=125
three\:AlertPosX=0
three\:AlertPosY=0
three\:AutoRun=0
three\:LoadFile=C:\\Users\\fcgle\\Downloads\\dialogue_demo.js
seconname\:Parked=1
seconname\:ConsolePosX=14
seconname\:ConsolePosY=18
seconname\:ConsoleSizeX=130
seconname\:ConsoleSizeY=35
seconname\:DialogPosX=375
seconname\:DialogPosY=250
seconname\:AlertPosX=225
seconname\:AlertPosY=50
seconname\:AutoRun=0
seconname\:LoadFile=
another\:Parked=0
another\:ConsolePosX=50
another\:ConsolePosY=561
another\:ConsoleSizeX=583
another\:ConsoleSizeY=297
another\:DialogPosX=375
another\:DialogPosY=250
another\:AlertPosX=225
another\:AlertPosY=50
another\:AutoRun=0
another\:LoadFile=
third\:Parked=1
third\:ConsolePosX=148
third\:ConsolePosY=18
third\:ConsoleSizeX=94
third\:ConsoleSizeY=35
third\:DialogPosX=375
third\:DialogPosY=250
third\:AlertPosX=225
third\:AlertPosY=50
third\:AutoRun=0
third\:LoadFile=
Consoles=three:seconname:another:third
[Directories]

rgleason, 2023-03-31:
Restart OCPN and activate the plugin
You mean to restart O and toggle JS on from the toolbar?
(...not to enable/disable).

rgleason, 2023-03-31:
Opened with notepad++ after Opencpn has been opened and JS toolbar button turned "ON"
It appears that the values are not being read or not being acted upon.

[PlugIns/JavaScript_pi]
VersionMajor=2
VersionMinor=1
ShowJavaScriptIcon=1
CurrentDirectory=C:\\Users\\fcgle\\OneDrive\\Documents
ShowingConsoles=0
Recents=C:\\Users\\fcgle\\Downloads\\another-demo.js;C:\\Users\\fcgle\\Downloads\\second-dialog-demo.js
Favourites=C:\\Users\\fcgle\\Downloads\\dialogue_demo.js;C:\\Users\\fcgle\\Downloads\\yet-another-demo.js
ParkingBespoke=1
ParkingMinHeight=35
ParkingStub=66
ParkingLevel=44
ParkingFirstX=36
ParkingSep=10
three\:Parked=0
three\:ConsolePosX=831
three\:ConsolePosY=486
three\:ConsoleSizeX=706
three\:ConsoleSizeY=366
three\:DialogPosX=630
three\:DialogPosY=125
three\:AlertPosX=0
three\:AlertPosY=0
three\:AutoRun=0
three\:LoadFile=C:\\Users\\fcgle\\Downloads\\dialogue_demo.js
seconname\:Parked=1
seconname\:ConsolePosX=14
seconname\:ConsolePosY=18
seconname\:ConsoleSizeX=130
seconname\:ConsoleSizeY=35
seconname\:DialogPosX=375
seconname\:DialogPosY=250
seconname\:AlertPosX=225
seconname\:AlertPosY=50
seconname\:AutoRun=0
seconname\:LoadFile=
another\:Parked=0
another\:ConsolePosX=50
another\:ConsolePosY=561
another\:ConsoleSizeX=583
another\:ConsoleSizeY=297
another\:DialogPosX=375
another\:DialogPosY=250
another\:AlertPosX=225
another\:AlertPosY=50
another\:AutoRun=0
another\:LoadFile=
third\:Parked=1
third\:ConsolePosX=148
third\:ConsolePosY=18
third\:ConsoleSizeX=94
third\:ConsoleSizeY=35
third\:DialogPosX=375
third\:DialogPosY=250
third\:AlertPosX=225
third\:AlertPosY=50
third\:AutoRun=0
third\:LoadFile=
Consoles=three:seconname:another:third

antipole2, 2023-03-31:
Rick... please test alert and dialogue positioning.

1. Run one line script `alert("This is an alert");`
2. Move alert before dismissing
3. Run same script in same window.  Alert should be in position from where it was dismissed.
4. Disable plugin and reenable it.
5. Run the script from **the same window**
6. Alert should be in the same position it last was in.

Do the same with a dialogue

Thanks

antipole2, 2023-03-31:
Fixes in the next build.
Adjustments were in the constructor but that does not seem to be working.
Have forced position after construction.
Also corrections for reconstructing a parked console.

rgleason, 2023-03-31:
Sorry getting ready for heat pumps...
Alert works as you said.
Same with dialogue.

antipole2, 2023-04-01:
Rick... fixes in build f8515c4.  Please try.

rgleason, 2023-04-01:
Installed.

[image]

JavaScript plugin version 2.1
JavaScript patch 3
JavaScript tools window DPI scaling factor 2.500000
wxWidgets version 3.2.1.0 or 3201
OCPN API version 1.18
Duktape version 20700
Not using svg
pJavaScript_pi->m_pconfig			0x00016b73a8
pJavaScript_pi->m_parent_window		0x0003cda940
favouriteFiles:
	C:\Users\fcgle\Downloads\dialogue_demo.js
	C:\Users\fcgle\Downloads\yet-another-demo.js
recentFiles:
	C:\Users\fcgle\Downloads\another-demo.js
	C:\Users\fcgle\Downloads\second-dialog-demo.js
pJavaScript_pi->mpFirstConsole	0x001c08f688

----------Console three----------
Console name:		three
this:					0x001c08f688
mpNextConsole:		0x001c2cd570
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	-173815440
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:35 y:45	DIP x:14 y:18
Size():				x:391 y:61	DIP	x:156 y:24	MinSize() x:241 y:87
m_parked:			true position x:14 y:18
isParked():			true
Messages callback table
	OCPN_CORE_SIGNALK		
	OCPN_OPENGL_CONFIG		
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------Console seconname----------
Console name:		seconname
this:					0x001c2cd570
mpNextConsole:		0x001c2ce8b0
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	1936028192
mWaitingCached:		true
[... 243 lines cut ...]
Messages callback table
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------------------------------------------------

pJavaScript_pi->mpBin		nullptr

End of dump

rgleason, 2023-04-01:
[image]

JavaScript plugin version 2.1
JavaScript patch 3
JavaScript tools window DPI scaling factor 2.500000
wxWidgets version 3.2.1.0 or 3201
OCPN API version 1.18
Duktape version 20700
Not using svg
pJavaScript_pi->m_pconfig			0x00016b73a8
pJavaScript_pi->m_parent_window		0x0003cda940
favouriteFiles:
	C:\Users\fcgle\Downloads\dialogue_demo.js
	C:\Users\fcgle\Downloads\yet-another-demo.js
recentFiles:
	C:\Users\fcgle\Downloads\another-demo.js
	C:\Users\fcgle\Downloads\second-dialog-demo.js
pJavaScript_pi->mpFirstConsole	0x001c08f688

----------Console three----------
Console name:		three
this:					0x001c08f688
mpNextConsole:		0x001c2cd570
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	-173815440
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:35 y:45	DIP x:14 y:18
Size():				x:391 y:61	DIP	x:156 y:24	MinSize() x:241 y:87
m_parked:			true position x:14 y:18
isParked():			true
Messages callback table
	OCPN_CORE_SIGNALK		
	OCPN_OPENGL_CONFIG		
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------Console seconname----------
Console name:		seconname
this:					0x001c2cd570
mpNextConsole:		0x001c2ce8b0
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	1936028192
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
[... 241 lines cut ...]
Messages callback table
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------------------------------------------------

pJavaScript_pi->mpBin		nullptr

End of dump

rgleason, 2023-04-01:
Tony, it is not remembering locations to begin with.
How would it remember locations when I turn off and disable JS?

[image]

[image]

JavaScript plugin version 2.1
JavaScript patch 3
JavaScript tools window DPI scaling factor 2.500000
wxWidgets version 3.2.1.0 or 3201
OCPN API version 1.18
Duktape version 20700
Not using svg
pJavaScript_pi->m_pconfig			0x00016b73a8
pJavaScript_pi->m_parent_window		0x0003cda940
favouriteFiles:
	C:\Users\fcgle\Downloads\dialogue_demo.js
	C:\Users\fcgle\Downloads\yet-another-demo.js
recentFiles:
	C:\Users\fcgle\Downloads\another-demo.js
	C:\Users\fcgle\Downloads\second-dialog-demo.js
pJavaScript_pi->mpFirstConsole	0x001c08f688

----------Console three----------
Console name:		three
this:					0x001c08f688
mpNextConsole:		0x001c2cd570
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	-173815440
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:35 y:45	DIP x:14 y:18
Size():				x:391 y:61	DIP	x:156 y:24	MinSize() x:241 y:87
m_parked:			true position x:14 y:18
isParked():			true
Messages callback table
	OCPN_CORE_SIGNALK		
	OCPN_OPENGL_CONFIG		
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------Console seconname----------
Console name:		seconname
this:					0x001c2cd570
mpNextConsole:		0x001c2ce8b0
mStatus:				(no status)
mRunningMain:		false
[... 246 lines cut ...]
Messages callback table
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------------------------------------------------

pJavaScript_pi->mpBin		nullptr

End of dump

antipole2, 2023-04-01:
Don't understand your comment " it is not remembering locations to begin with."
Not sure which screenshot is which.  If the last two were before and after disabling, then it looks like it is working.
Don't bother with dumps until I know there is still an issue.

rgleason, 2023-04-01:
Tony, I went to close JS at the toolbar when I hit JS, the old consoles appeared in the upper left all stacked up.  I messed around with those and they were not expanding to the old places.
Then I turned off JS at the toolbar and disabled JS at the PIM list.
Then renabled and the old consoles appeared stacked up in one place.
[image]

rgleason, 2023-04-01:
Then unstacked them and both sets were there.
[image]

Then expanded and minimized and the original set went back into the expected parking place.
[image]

rgleason, 2023-04-01:
Then tried hitting the X to expand on two of them and they did not land where it was expected earlier.

[image]

[image]

[image]

JavaScript plugin version 2.1
JavaScript patch 3
JavaScript tools window DPI scaling factor 2.500000
wxWidgets version 3.2.1.0 or 3201
OCPN API version 1.18
Duktape version 20700
Not using svg
pJavaScript_pi->m_pconfig			0x00016b73a8
pJavaScript_pi->m_parent_window		0x0003cda940
favouriteFiles:
	C:\Users\fcgle\Downloads\dialogue_demo.js
	C:\Users\fcgle\Downloads\yet-another-demo.js
recentFiles:
	C:\Users\fcgle\Downloads\another-demo.js
	C:\Users\fcgle\Downloads\second-dialog-demo.js
pJavaScript_pi->mpFirstConsole	0x001c090200

----------Console three----------
Console name:		three
this:					0x001c090200
mpNextConsole:		0x001a2ac968
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	1633841761
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:36 y:44	DIP x:14 y:18
Size():				x:241 y:88	DIP	x:96 y:35	MinSize() x:241 y:87
m_parked:			true position x:14 y:18
isParked():			true
Messages callback table
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------Console seconname----------
Console name:		seconname
this:					0x001a2ac968
mpNextConsole:		0x001c08f688
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
[... 239 lines cut ...]
Messages callback table
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------------------------------------------------

pJavaScript_pi->mpBin		nullptr

End of dump

rgleason, 2023-04-01:
I also tried expanding the new set of 4 and they did not remember where they were, but just used the current location to expand from the top left corner.
When I then shrank them, they popped back to the parking position at the end of the line.

antipole2, 2023-04-01:
Let's slow down a bit and no dumps please unless I request them.

1) Toggling the toolbar only hides and shows - it does not move anything, so unhiding will show them exactly as they were before.
2) When you park a console it is minimised and moved to its parking place.  If you then hit the X it expands to show the message that you cannot close it unless cleared.  It expands in place and does not move from it sparking place.  It does not move back to where it was parked from.

antipole2, 2023-04-01:
Let's try just four consoles
2  parked and 2 elsewhere and not minimised.
Screen shot 1
Then disable plugin in the Settings and re-enable.
Screennshot 2

rgleason, 2023-04-01:
RE: " It expands in place and does not move from it sparking place. It does not move back to where it was parked from."
I think I experienced something else, earlier. If you expanded it it moved to where it was parked. Or am I confusing something here?

Screen shot 1
[image]

Screen shot2
[image]

[image]

[image]

antipole2, 2023-04-01:
I don't understand what you are showing me here.
Perhaps you need to spell out what each shot is showing?
We need to focus on: does the console get put back where it was and with the same size after deactivating and reactivating the plugin?

rgleason, 2023-04-02:
Tony,

No the console does not get put back where it was after I disable JS and then enable JS. The two open consoles are relocated and resized now.  See Screen Shot 1 and Screenshot 2.

I tried this again this morning with the same result.

Sorry.

antipole2, 2023-04-02:
I cannot tie the previous dumps to the screenshots, so o ask you to go through it once again:

Just four consoles
2 parked and 2 elsewhere and not minimised.

1. Screen shot 1
2. Console dump 1
3. Close OCPN
4. Copy JS section of .ini file
5. Restart OCPN and toggle JS if needed to see consoles
o one package if you wish - making sure the file names identify which is which6. Screenshot 2
7. Console dump 2

Please take care not to touch or move the consoles once the sequence is started.
Suggest you zip the 5 things into a single file to stop this issue getting so long.  As long as the file names make  clear which is what.
Thanks for your forbearance.

rgleason, 2023-04-02:
1.
[image]
2. JavaScript plugin version 2.1
JavaScript patch 3
JavaScript tools window DPI scaling factor 2.500000
wxWidgets version 3.2.1.0 or 3201
OCPN API version 1.18
Duktape version 20700
Not using svg
pJavaScript_pi->m_pconfig			0x0000add3a0
pJavaScript_pi->m_parent_window		0x0003a1d610
favouriteFiles:
	C:\Users\fcgle\Downloads\dialogue_demo.js
	C:\Users\fcgle\Downloads\yet-another-demo.js
recentFiles:
	C:\Users\fcgle\Downloads\another-demo.js
	C:\Users\fcgle\Downloads\second-dialog-demo.js
pJavaScript_pi->mpFirstConsole	0x001ba8d6d8

----------Console test1----------
Console name:		test1
this:					0x001ba8d6d8
mpNextConsole:		0x001bb974e8
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	0
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:2477 y:1586	DIP x:991 y:634
Size():				x:1365 y:554	DIP	x:546 y:222	MinSize() x:237 y:87
m_parked:			true position x:306 y:18
isParked():			false
Messages callback table
	OCPN_CORE_SIGNALK		
	OCPN_OPENGL_CONFIG		
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------Console test2----------
Console name:		test2
this:					0x001bb974e8
mpNextConsole:		0x001bbbe620
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	16777472
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:2149 y:189	DIP x:860 y:76
Size():				x:1700 y:608	DIP	x:680 y:243	MinSize() x:237 y:87
m_parked:			true position x:208 y:18
isParked():			false
Messages callback table
	OCPN_CORE_SIGNALK		
	OCPN_OPENGL_CONFIG		
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------Console test3----------
Console name:		test3
this:					0x001bbbe620
mpNextConsole:		0x001bbd7218
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	0
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:272 y:44	DIP x:109 y:18
Size():				x:237 y:88	DIP	x:95 y:35	MinSize() x:237 y:87
m_parked:			true position x:109 y:18
isParked():			true
Messages callback table
	OCPN_CORE_SIGNALK		
	OCPN_OPENGL_CONFIG		
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------Console four----------
Console name:		four
this:					0x001bbd7218
mpNextConsole:		nullptr
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	2097184
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:36 y:44	DIP x:14 y:18
Size():				x:226 y:88	DIP	x:90 y:35	MinSize() x:226 y:87
m_parked:			true position x:14 y:18
isParked():			true
Messages callback table
	OCPN_CORE_SIGNALK		
	OCPN_OPENGL_CONFIG		
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------------------------------------------------

pJavaScript_pi->mpBin		nullptr

End of dump

rgleason, 2023-04-02:
3. Close OpenCPN
4. Copy JS section of ini

[PlugIns/JavaScript_pi]
VersionMajor=2
VersionMinor=1
ShowJavaScriptIcon=1
CurrentDirectory=C:\\Users\\fcgle\\OneDrive\\Documents
ShowingConsoles=1
Recents=C:\\Users\\fcgle\\Downloads\\another-demo.js;C:\\Users\\fcgle\\Downloads\\second-dialog-demo.js
Favourites=C:\\Users\\fcgle\\Downloads\\dialogue_demo.js;C:\\Users\\fcgle\\Downloads\\yet-another-demo.js
ParkingBespoke=1
ParkingMinHeight=35
ParkingStub=66
ParkingLevel=44
ParkingFirstX=36
ParkingSep=10
test1\:Parked=0
test1\:ConsolePosX=991
test1\:ConsolePosY=634
test1\:ConsoleSizeX=546
test1\:ConsoleSizeY=222
test1\:DialogPosX=375
test1\:DialogPosY=250
test1\:AlertPosX=225
test1\:AlertPosY=50
test1\:AutoRun=0
test1\:LoadFile=
test2\:Parked=0
test2\:ConsolePosX=860
test2\:ConsolePosY=76
test2\:ConsoleSizeX=680
test2\:ConsoleSizeY=243
test2\:DialogPosX=375
test2\:DialogPosY=250
test2\:AlertPosX=225
test2\:AlertPosY=50
test2\:AutoRun=0
test2\:LoadFile=
test3\:Parked=1
test3\:ConsolePosX=109
test3\:ConsolePosY=18
test3\:ConsoleSizeX=95
test3\:ConsoleSizeY=35
test3\:DialogPosX=375
test3\:DialogPosY=250
test3\:AlertPosX=225
test3\:AlertPosY=50
test3\:AutoRun=0
test3\:LoadFile=
four\:Parked=1
four\:ConsolePosX=14
four\:ConsolePosY=18
four\:ConsoleSizeX=90
four\:ConsoleSizeY=35
four\:DialogPosX=375
four\:DialogPosY=250
four\:AlertPosX=225
four\:AlertPosY=50
four\:AutoRun=0
four\:LoadFile=
Consoles=test1:test2:test3:four
[Directories]

rgleason, 2023-04-02:
Restart OCPN and toggle JS if needed to see consoles
o one package if you wish - making sure the file names identify which is which6. Screenshot 2

[image]

Console dump 2

JavaScript plugin version 2.1
JavaScript patch 3
JavaScript tools window DPI scaling factor 2.500000
wxWidgets version 3.2.1.0 or 3201
OCPN API version 1.18
Duktape version 20700
Not using svg
pJavaScript_pi->m_pconfig			0x0001a611e8
pJavaScript_pi->m_parent_window		0x00098dc780
favouriteFiles:
	C:\Users\fcgle\Downloads\dialogue_demo.js
	C:\Users\fcgle\Downloads\yet-another-demo.js
recentFiles:
	C:\Users\fcgle\Downloads\another-demo.js
	C:\Users\fcgle\Downloads\second-dialog-demo.js
pJavaScript_pi->mpFirstConsole	0x001c4b47f0

----------Console test1----------
Console name:		test1
this:					0x001c4b47f0
mpNextConsole:		0x001c88a848
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	-229
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:85 y:160	DIP x:34 y:64
Size():				x:1397 y:643	DIP	x:559 y:257	MinSize() x:237 y:87
m_parked:			false position x:0 y:0
isParked():			false
Messages callback table
	OCPN_CORE_SIGNALK		
	OCPN_OPENGL_CONFIG		
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------Console test2----------
Console name:		test2
this:					0x001c88a848
mpNextConsole:		0x001c8a5838
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	16777472
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:85 y:160	DIP x:34 y:64
Size():				x:1732 y:696	DIP	x:693 y:278	MinSize() x:237 y:87
m_parked:			false position x:0 y:0
isParked():			false
Messages callback table
	OCPN_CORE_SIGNALK		
	OCPN_OPENGL_CONFIG		
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------Console test3----------
Console name:		test3
this:					0x001c8a5838
mpNextConsole:		0x001c8a7d90
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	7274596
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:85 y:160	DIP x:34 y:64
Size():				x:391 y:176	DIP	x:156 y:70	MinSize() x:237 y:87
m_parked:			true position x:109 y:18
isParked():			false
Messages callback table
	OCPN_CORE_SIGNALK		
	OCPN_OPENGL_CONFIG		
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------Console four----------
Console name:		four
this:					0x001c8a7d90
mpNextConsole:		nullptr
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	7274596
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:85 y:160	DIP x:34 y:64
Size():				x:391 y:176	DIP	x:156 y:70	MinSize() x:226 y:87
m_parked:			true position x:14 y:18
isParked():			false
Messages callback table
	OCPN_CORE_SIGNALK		
	OCPN_OPENGL_CONFIG		
	OpenCPN Config		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------------------------------------------------

pJavaScript_pi->mpBin		nullptr

End of dump

antipole2, 2023-04-02:
Stumped on why it is not working.  Have asked Mike to check it out on a standard display in case it is a Windows issue.
Meanwhile, please attend to [Dialogue](https://github.com/antipole2/JavaScript_pi/issues/64#issuecomment-1492335573) and [Tools Consoles tab (only)](https://github.com/antipole2/JavaScript_pi/issues/62#issuecomment-1492988995).

antipole2, 2023-04-02:
Mike Rossiter
[20:01](https://opencpn.zulipchat.com/#narrow/pm-with/399559,554731-pm/near/346402608)
We never solved the position problem for Windows. I will take a look tomorrow. s/v Kamish gets launched tomorrow so I am off to the river on Tuesday.

rgleason, 2023-04-03:
Beginning the season! Exciting.

antipole2, 2023-04-03:
Build 7726527 is another go at the console positioning thing.
Please check if, after disabling and reenabling in the Settings window, the consoles get their position back for all end up against the left side?
Don't need dumps at the moment - just to know whether they are still jammed up against the left side.

Another test, please.  With this build, I am doing a test...  If you have a console over the canvas or partially over it, if you click on the canvas or hover over the Toolbar, does the console disappear behind the canvas?  With this test build, on MacOS the window goes behind the canvas.

antipole2, 2023-04-21:
Thinned out obsolete images and dumps.
All done as far as I know.

closed 2023-04-21 by antipole2