antipole2/JavaScript_pi #78: Sending Signal K

antipole2, 2023-10-17:
It is desirable that the JavaScript plugin can be used to work with Signal K. Given the close relationship of both Signal K and JavaScript with JSON, this should be relatively easy to do and to use.

I do not have a Signal K set up and am away from my boat for the winter. So I am hoping SKS will work with me to establish this capability.

**Sending Signal K messages**
This should be possible using WriteCommDriver but needs testing by a SignalK user.

antipole2, 2024-01-27:
@barcoMeCasa / Paddy.. please run this script and report output

writeCommDriver not yet implemented.  That is what we will be working on but probably not that name.
Please run the following script and report output:
```
handles = OCPNgetActiveDriverHandles();
for (h = 0; h < handles.length; h++){
	attributes = OCPNgetDriverAttributes(handles[h]);
	print(handles[h], "\n\t",attributes, "\n");
	}
```

boatybits, 2024-01-27:
getting an error with that >
[image]

From the diagnostic dump > 

```
JavaScript plugin version 2.1
JavaScript patch 6
JavaScript tools window DPI scaling factor 1.500000
wxWidgets version 3.2.1.0 or 3201
OCPN API version 1.18
Duktape version 20700
Not using svg
pJavaScript_pi->m_pconfig					0x0000dbc1c0
pJavaScript_pi->m_parent_window			0x00041bc270
pJavaScript_pi->m_floatOnTop				false
pJavaScript_pi->mRememberToggleStatus	false
favouriteFiles:
recentFiles:
pJavaScript_pi->mpFirstConsole	0x000a449690

----------Console JavaScript----------
Console name:		JavaScript
this:					0x000a449690
mpNextConsole:		nullptr
mStatus:				HAD_ERROR 
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	1
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:				false
position:				screen x:497 y:95	DIP x:331 y:63
Size():				x:1377 y:943	DIP	x:918 y:629	MinSize() x:199 y:56
m_parked:			false position x:0 y:0
isParked():			false
Messages callback table
	LOGBOOK_READY_FOR_REQUESTS		
	LOGBOOK_WINDOW_HIDDEN		
	OCPN_CORE_SIGNALK		
	OCPN_DRAW_PI_READY_FOR_REQUESTS		
	OCPN_OPENGL_CONFIG		
	OCPN_TRK_ACTIVATED		
	OCPN_TRK_DEACTIVATED		
	OpenCPN Config		
	PYPILOT_HOST		
	PYPILOT_HOST_REQUEST		
	WMM_VARIATION_BOAT		
	WMM_VARIATION_BOAT_REQUEST		
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_activeLegFunction:		
m_exitFunction:	
m_explicitResult:		false
m_result:				ReferenceError: identifier 'OCPNgetActiveDriverHandles' undefined
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
No Duktape context

----------------------------------------------------

pJavaScript_pi->mpBin		nullptr
```

antipole2, 2024-01-27:
@boatybits This is testing v3.0 Beta - not valid for earlier releases.  If you want to play, you need the [latest beta release from here](https://cloudsmith.io/~opencpn/repos/javascript-beta/packages/). [ User guide here](https://github.com/antipole2/JavaScript_pi/blob/v3.0.0-beta/documentation/JavaScript_plugin_user_guide.pdf).  This is work in progress!
This present thread for Sending Signal K.
For more general matters, there is a [new discussion forum here](https://github.com/antipole2/JavaScript_pi/discussions).

boatybits, 2024-01-27:
That's where I got the tarball, think I'm on the latest beta , javascript-v3.0.0-beta-a_msvc-wx32-10-win32.tar.gz
I'll try again

antipole2, 2024-01-28:
Ready to test sending SignalK.

You will need to download build _beta b_ (or later) from my [beta repository](https://cloudsmith.io/~opencpn/repos/javascript-beta/packages/).
Attached is a starter test script.  This
1. locates the handle for the SignalK port
2. builds a SignalK object
3. converts it to JSON
4. writes it out using a new `OCPNpushText(text, handle)` API

The object built in my attached script is illustrative only - it presumably does not make SignalK sense.  But it illustrates how you could do it.  I have not been able to test the new `OCPNpushText(text, handle)` API for SignalK but if pushing through an NMEA0183 handle, it works correctly and does the same as `OCPNpushNMEA0183(text, handle)`.

When sharing output or scripts, please paste text rather than screen-shotting, unless that is needed to show a particular thing.
[writeSignalK.js.zip](https://github.com/antipole2/JavaScript_pi/files/14076045/writeSignalK.js.zip)

boatybits, 2024-01-28:
plugin updated ok but the script crashes opencpn unless I comment out OCPNpushText(skJson, handle);
Don't see anything in opencpn log, it looks the same as it does if booted up but the script not run.

antipole2, 2024-01-28:
> plugin updated ok but the script crashes opencpn unless I comment out OCPNpushText(skJson, handle);
> 
> Don't see anything in opencpn log, it looks the same as it does if booted up but the script not run. 

Can you let me have the crash report, please?

boatybits, 2024-01-29:
`<?xml version="1.0" encoding="UTF-8" ?>
<CrashRpt version="1403">
    <CrashGUID>77585162-3bc9-4900-8e47-ee6b70ac4ad0</CrashGUID>
    <AppName>OpenCPN</AppName>
    <AppVersion>5.8.0.0</AppVersion>
    <ImageName>C:\Program Files (x86)\OpenCPN\opencpn.exe</ImageName>
    <OperatingSystem>Windows 10 Enterprise Build 19045</OperatingSystem>
    <OSIs64Bit>1</OSIs64Bit>
    <GeoLocation>en-gb</GeoLocation>
    <SystemTimeUTC>2024-01-27T11:52:13Z</SystemTimeUTC>
    <ExceptionAddress>0x68040b17</ExceptionAddress>
    <ExceptionModule>C:\Users\Padz\AppData\Local\opencpn\plugins\JavaScript_pi.dll</ExceptionModule>
    <ExceptionModuleBase>0x68010000</ExceptionModuleBase>
    <ExceptionModuleVersion>0.0.0.0</ExceptionModuleVersion>
    <ExceptionType>0</ExceptionType>
    <ExceptionCode>0</ExceptionCode>
    <GUIResourceCount>2314</GUIResourceCount>
    <OpenHandleCount>572</OpenHandleCount>
    <MemoryUsageKbytes>245508</MemoryUsageKbytes>
    <ScreenshotInfo>
        <VirtualScreen left="0" top="0" width="1280" height="720" />
        <Monitors>
            <Monitor left="0" top="0" width="1280" height="720" file="screenshot0.jpg" />
        </Monitors>
        <Windows>
            <Window left="409" top="168" width="492" height="400" title="Test2" />
            <Window left="331" top="63" width="680" height="700" title="JavaScript" />
            <Window left="680" top="0" width="600" height="637" title="" />
            <Window left="3" top="47" width="79" height="585" title="" />
            <Window left="1065" top="654" width="213" height="39" title="" />
            <Window left="85" top="478" width="115" height="177" title="nav" />
            <Window left="85" top="61" width="137" height="409" title="wind" />
            <Window left="-7" top="-7" width="1295" height="735" title="OpenCPN 5.8.4-0" />
        </Windows>
    </ScreenshotInfo>
    <CustomProps />
    <FileList>
        <FileItem name="crashdump.dmp" description="Crash Minidump" />
        <FileItem name="crashrpt.xml" description="Crash Description XML" />
        <FileItem name="opencpn.ini" description="" optional="1" />
        <FileItem name="opencpn.log" description="" optional="1" />
        <FileItem name="screenshot0.jpg" description="Desktop Screenshot" />
    </FileList>
</CrashRpt>
`

antipole2, 2024-01-29:
Thanks - and your script, please zipped.

boatybits, 2024-01-29:
that was using your script from https://github.com/antipole2/JavaScript_pi/files/14076045/writeSignalK.js.zip
changing the last line to
//OCPNpushText(skJson, handle);
gave this output >
`{
	"self": "123456789",
	"vessels": {
		"name": "Titanic",
		"mmsi": 123456789,
		"navigation": {
			"headingTrue": {
				"value": 23,
				"source": "self",
				"timestamp": "2014-03-24T00:15:41Z"
			},
			"headingMagnetic": {
				"value": 43,
				"source": "self",
				"timestamp": "2014-03-24T00:15:41Z"
			}
		}
	}
}
result: {"self":"123456789","vessels":{"name":"Titanic","mmsi":123456789,"navigation":{"headingTrue":{"value":23,"source":"self","timestamp":"2014-03-24T00:15:41Z"},"headingMagnetic":{"value":43,"source":"self","timestamp":"2014-03-24T00:15:41Z"}}}}`

antipole2, 2024-01-29:
@boatybits I know that pushing NMEA0183 data works OK because I can test that.  I have now rationalised the push methods so OCPNpushText uses exactly the same code as OCPNpushNMEA0183, apart from the latter adding the checksum.
Please try again with new build beta-c.

The SignalK data in my script was illustrative only.  I do not know SignalK.  Possibly a problem?  Please replace with something you know is valid SignalK and appropriate to send.

If it crashes again, please let me have both the script and the crash report.

Thanks for your help with this.

boatybits, 2024-01-30:
Hi, must be getting close..still crashing though. 
Hopefully attached is zipped  crashdump folder & script. Same as before, works OK with last line commented out. 
Output looks OK, not sure how well you know signalk but there's a data fiddler page where you can test data, pasting in the output seems to be fine > 
[image]

And maybe this will be useful, I've a linux machine running in the cloud which I loaded signalk onto >
http://157.245.29.42:3000/
user & pass both "cloudop"
Nothing important on it so do anything you want & don't worry about breaking it. 

[sigk.zip](https://github.com/antipole2/JavaScript_pi/files/14104909/sigk.zip)

antipole2, 2024-01-31:
Going to need input from others here.  To document what I am doing, the crux is the following code:

```
// handle is the port handle, protocol is the protocol for that handle, data is the data text
	if (protocol == "nmea0183"){
		data = addNMEA0183CheckSum(ctx, data);
		}
	auto payload = make_shared<std::vector<uint8_t>>();
	for (const auto& ch : data) payload->push_back(ch);
	CommDriverResult outcome = WriteCommDriver(handle, payload);
```
If the protocol for the handle is for an nmea0183 network connection,  this works fine and the nmea0183 sentence is received by the other end OK.
If attempting to send to a SignalK port, we get a crash as reported above.

TwoCanPlugIn, 2024-02-02:
Hi Tony,

Can't say I've had a lot of experience with SignalK, and to be honest, I'm not terribly interested in it.

That being said, I just played with the following code in one of my test plugins, on Windows.

It doesn't work, but it doesn't crash OpenCPN!

`CommDriverResult result;`
`wxString message = "{\"updates\":[{\"source\":`
`{\"sentence\":\"HDM\",\"talker\":\"II\",\"type\":\"NMEA0183\"},\"values\":`
`[{\"path\":\"navigation.headingMagnetic\",\"value\":2}]}],\"context\":\"vessels.self\",\"name\":\"Iona\"}";`
`std::vector<uint8_t>SignalK;`
`for (auto it : message) {`
		    `SignalK.push_back(it);`
	`}`
	`auto signalkPointer = std::make_shared<std::vector<uint8_t> >(std::move(SignalK));`
	`result = WriteCommDriver(driverSignalK, signalkPointer);`

The return code is 2, which means invalid parameter. I'm pretty sure the driver handle is OK, as I re-use similar code for writing NMEA 2000 stuff.

So I think your crashing of OpenCPN is perhaps a bug in your code ?

Anyway, a few questions arise.

Firstly, does SignalK Server need to be configured for a SignalK incoming data connection, similar to how NMEA 2000 or NMEA 0183 connections are created ? This is different to configuring the SignalK HTTP port, or SignalK over TCP.

Secondly, do you have a JSON SignalK update message that is "known" to update a value in SignalK ? 

I can receive SignalK updates from a SignalK Server over TCP (port 8375) using netcat on Linux and a Powershell script on Windows, the obvious test is to use either of these to send an update to SignalK Server. I haven't been able to successfully do that.
If that can be proven, the next test is to send it over web sockets, which is the transport that OpenCPN uses.

Thirdly, we both know that GetSignalKPayload was not initially implemented in 5.8.x, who's to say that WriteCommDriver over SignalK has either been implemented or verified to work ?

Regards.

TwoCanPlugIn, 2024-02-12:
Hi Tony,

Have done some further testing with some Powershell scripts on Windows, just to verify that it is possible to update values on SignalK Server via web sockets. Now just a matter for you to get the same to work for your Javascript plugin.

This is the uri for my test setup. (I'm not worried about private IP addresses, usernames or passwords)

ws://192.168.43.170:3000/signalk/v1/stream/?subscribe=none

II haven't been able to get this to work with a "working" subscription, eg.
subscribe=all  or subscribe = {"context":"vessels.self","subscribe":[{"path":"environment.*"},{"path":"navigation.*"}]}

It only appears to work with subscribe=none

After connecting to the SignalK Server web socket with no subscription, you receive the server announcement:

{"name":"signalk-server","version":"2.1.0","self":"vessels.urn:mrn:signalk:uuid:1cb1a66a-814c-4478-8b84-701eec9524bb","roles":["master","main"],"timestamp":"2024-02-12T22:11:19.510Z"}

Then send the login request (On Windows I use the guidgen utility to create a valid GUID/UUID)

"{\"requestId\":\"FA1CA3B7-F121-4E5C-99FA-A498BD5CAFEB\",\"login\":{\"username\":\"pi\",\"password\":\"raspberry\"}}";

I don't recall any specific configuration of SignalK Server so I assume security is enabled by default

The successful login response is as follows:

{"requestId":"FA1CA3B7-F121-4E5C-99FA-A498BD5CAFEB","state":"COMPLETED","statusCode":200,"login":{"token":"eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpZCI6InBpIiwiaWF0IjoxNzA3Nzc1ODgzLCJleHAiOjE3MDc4NjIyODN9.RoVbGhAQZ5P7lfNIa0dSmQKZGsRa2naeTRjNd8XC-ck"}}

Then send the update message, in this case the value for heading in PGN 127250. I am not sure if any of the other SignalK stuff (label, type, context) or NMEA 2000 stuff (source or deviceInstance) is required.

"{\"updates\":[{\"source\":{\"label\":\"CAN-BUS\",\"type\":\"NMEA2000\",\"pgn\":127250,\"src\":\"2\",\"deviceInstance\":0},\"values\":[{\"path\":\"navigation.headingTrue\",\"value\":2.0}]}],\"context\":\"vessels.self\"}";

The successful update response is then received.

{"requestId":"FA1CA3B7-F121-4E5C-99FA-A498BD5CAFEB","state":"COMPLETED","statusCode":200,"login":{"token":"eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpZCI6InBpIiwiaWF0IjoxNzA3Nzc1ODg4LCJleHAiOjE3MDc4NjIyODh9.MAJarWkyAofdZ_-EV7do8Hnb7zUOGNJVaizN0EhqIzc"}}

The attached screenshot shows the updated value in the SignalK Server Data Browser.

I sincerely doubt whether WriteCommDriver for a SignalK connection has ever had a specification written, correctly implemented nor even tested. My gut feel is that when OpenCPN initiates a connection to SignalK Server, it does so with subscribe=all, so I doubt that it is possible to peform a write without doing a subscribe=none, perform the login & update and then doing a subscribe=all

As many of us are discovering, there are many features in OpenCPN that just don't work. I don't understand how the developers can check in a feature without performing a unit test to actually ensure it works as expected. I even doubt whether the SignalK Server discovery (msDNS/Bonjour) actually works (it has never for me),. One day I might just do a network capture to see if anything is even sent or received across the network.

Once you get your plugin and WriteCommDriver working without crashing OpenCPN, feel free to file the bug report.

[image]

TwoCanPlugIn, 2024-05-14:
I thought I would close the loop on this.

It is possible to update SignalK Server using web sockets from any other application, however at the moment writing to SignalK Server from OpenCPN is not supported. From plugin_api.cpp
```
if (protocol == "nmea0183") {
....
} else
    return RESULT_COMM_INVALID_PARMS;
}
```