antipole2/JavaScript_pi #88: messageBox cancel does not throw error

antipole2, 2024-04-06:
The User Guided states that if you cancel a messagBox dialogue, an error is thrown.

In v3.0 It appears to instead be returning a code of 5101.

This needs resolving.

antipole2, 2024-12-11:
This and associated code has had to be reworked to cope with Windows oddities.  An error was introduced into plugin v3 with this  but has now been fixed.

Run the one line script
`messageBox("Hello Worldl!");`
If you dismiss this with OK, it returns 1
If you cancel the box, it now throws an error correctly and as documented

torrmundi, 2024-12-11:
I got these results for terminating the messageBox with OK, then with Cancel:
result: 1
line 1 Error: Console Testing - messageBox cancelled

torrmundi, 2024-12-11:
The Fibonacci code result may have an issue:
"Fibonacci says: 0 1 1 2 3 5 8 13 21 34 55 89 144 233 377 610 987 1597 2584 4181
result: undefined"

This code:
```
const elements = ['Fire', 'Air', 'Water'];
elements.join();
```
returns "result: Fire,Air,Water"

antipole2, 2024-12-12:
What's wrong with your. Fibonacci test?

NB If you are running the example script from the User Guide p7, then the script 'result' is undefined - see page 15 "Understanding the result"

> For
> print("Hi there!\n");
> this will display  Hi there! and the result is undefined as the print function does not return a result.

torrmundi, 2024-12-13:
yes, undefined result, which I thought might be an error.
Thanks!