antipole2/JavaScript_pi #90: Windows download for v3.0 Build 1e Apr  1 2024 from catalogue has checksum error

antipole2, 2024-04-07:
@bcn @seeUlater
[See discussion here](https://www.cruisersforum.com/forums/f134/vdr-plugin-59808-10.html#post3888299).

Don't understand how this got through the checks.  Have asked Dave's advice.

bdbcat, 2024-04-08:
Usually this happens because the CI build process has been repeated, with no real code change.  I think...
Best solution:  Bump the version number, and resubmit the CI build

Rasbats, 2024-04-08:
On Windows with OpenCPN 5.9.x I did not have a problem.

antipole2, 2024-04-08:
v3.0.1 now in catalogue.
Please can SKS check if it installs on Windows this time?

antipole2, 2024-04-08:
I hear that v3.0.1 can be installed through the catalogue on Windows.

antipole2, 2024-04-08:
losing as issue resolved with new build

closed 2024-04-08 by antipole2