antipole2/JavaScript_pi #91: Plugin version available for Windows 8 not current

antipole2, 2024-04-08:
On a computer running Windows 8 with OpenCPN 5.8.4, the Javascript version available in the plugin catalogs (master, beta or alpha) is named -1.0.0. 

Submitted by @Erickalon

antipole2, 2024-04-08:
Please let me have a copy of the initial script in a new console or its output when run.  It looks something like
`Hello from the JavaScript plugin v3.0.1 Patch 1a Apr  8 2024 JS v20700`

Also please open the tools diagnostics tab and click on Dump and let me have the first 20 lines of the dumped text.

Erickalon, 2024-04-08:
Initial script in a new console:
print("Hello from the JavaScript plugin v3.0.1 Patch 1a Apr  8 2024 JS v20700\n");
"All OK";
Dumped text:
JavaScript plugin version 3.0
JavaScript patch 1
JavaScript tools window DPI scaling factor 1.000000
Platform	Linux 6.6.20+rpt-rpi-v8 aarch64
wxWidgets version 3.2.2.0 or 3202
OCPN API version 1.18
Duktape version 20700
Not using svg
pJavaScript_pi->m_pconfig					0x00bba8c120
pJavaScript_pi->m_parent_window			0x00bc346800
pJavaScript_pi->m_floatOnTop				false
pJavaScript_pi->mRememberToggleStatus	false
favouriteFiles:JavaScript plugin version 3.0
JavaScript patch 1
JavaScript tools window DPI scaling factor 1.000000
Platform	Linux 6.6.20+rpt-rpi-v8 aarch64
wxWidgets version 3.2.2.0 or 3202
OCPN API version 1.18
Duktape version 20700
Not using svg
pJavaScript_pi->m_pconfig					0x00bba8c120
pJavaScript_pi->m_parent_window			0x00bc346800
pJavaScript_pi->m_floatOnTop				false
pJavaScript_pi->mRememberToggleStatus	false
favouriteFiles:
recentFiles:
N2K pgn registrations: none
pJavaScript_pi->mpFirstConsole	0x00bfba1a40

----------Console JavaScript----------
Console name:		JavaScript
this:					0x00bfba1a40
mpNextConsole:		nullptr
mStatus:				(no status)

recentFiles:
N2K pgn registrations: none
pJavaScript_pi->mpFirstConsole	0x00bfba1a40

----------Console JavaScript----------
Console name:		JavaScript
this:					0x00bfba1a40
mpNextConsole:		nullptr
mStatus:				(no status)

antipole2, 2024-04-08:
@bdbcat 
[Erickalon](https://github.com/Erickalon)reports that on his Windows 8 with OpenCPN 5.8.4, the Javascript version available in the plugin catalogs (master, beta or alpha) is named -1.0.0 and not v3.0.1a.

The above test and dump confirm he is actually loading v3.0.1a but the catalogue is showing it as 1.0.0.
I did have a mis-build a few days ago that was tagged 1.0.0 but I have removed all occurrences out of the metadata before building v3.0.1a.

This is not very serious in that he is installing the latest build but it might point to an issue that needs attention?

antipole2, 2024-04-20:
This issue arises when the tag declaring the version number is of the form v3-2-0 rather than 3.2.0.
I have patched the metadata files to use the dot notation and it all looks correct to me now.

Can we close this?

bdbcat, 2024-04-21:
Yes, ready to close.

closed 2024-04-21 by antipole2