antipole2/JavaScript_pi #99: JavaScript build_race_courses.js crashes opencpn

antipole2, 2024-04-27:
### Discussed in https://github.com/antipole2/JavaScript_pi/discussions/98

<div type='discussions-op-text'>

<sup>Originally posted by **LordLaingholm** April 27, 2024</sup>
Hi all, 

I've recently installed the JavaScript plugin and loaded the build_race_courses.js script however when I run the script OpenCPN crashes immediately. I tailed the opencpn.log file at the time but no data was added to it. The JS console remains open. I've repeated the same exercise several times with the same result. Other javascripts run fine.

I'm running OpenCPN 5.8.4-1 on Linux openplotter 5.15.76-v8+, Raspberry Pi 4, I'm using VNC to access the pi.  
pi@openplotter:~ $ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 11 (bullseye)"
NAME="Debian GNU/Linux"
VERSION_ID="11"
VERSION="11 (bullseye)"
VERSION_CODENAME=bullseye
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

I'm new to javascript and the js plugin however I wrote a couple of scripts that worked fine.

This is the plugin diagnostic dump
JavaScript plugin version 3.0
JavaScript patch 0
JavaScript tools window DPI scaling factor 1.000000
Platform	Linux 5.15.76-v8+ aarch64
wxWidgets version 3.2.1.0 or 3201
OCPN API version 1.18
Duktape version 20700
Not using svg
pJavaScript_pi->m_pconfig					0x00b2dc9d20
pJavaScript_pi->m_parent_window			0x00b34bbb90
pJavaScript_pi->m_floatOnTop				false
pJavaScript_pi->mRememberToggleStatus	false
favouriteFiles:
recentFiles:
N2K pgn registrations: none
pJavaScript_pi->mpFirstConsole	0x00b672b690

----------Console JavaScript----------
Console name:		JavaScript
this:					0x00b672b690
mpNextConsole:		nullptr
mStatus:				(no status)
mRunningMain:		false
m_time_to_allocate:	1000ms
m_timeout_check_counter:	0
mWaitingCached:		true
mWaiting:			false
mpMessageDialog:	nullptr
isBusy():				false
isWaiting():			false
auto_run:			false
position:				screen x:-2 y:-2	DIP x:-2 y:-2
Size():				x:1024 y:740	DIP	x:1024 y:740	MinSize() x:123 y:30
m_parked:			true
isParked():			false
parked Location		100	27	size 125	31
notParked location	132	66	size 738	600
Messages callback table
	OCPN_CORE_SIGNALK
	OCPN_OPENGL_CONFIG
	OpenCPN Config
m_timerActionBusy:	false
Timers callback table
	(empty)
Menus callback table
	(empty)
m_dialog:			None
m_alert:				None
m_NMEAmessageFunction:		
m_streamMessageCntlsVector	<empty>
m_activeLegFunction:		
m_exitFunction:	
m_closeButtonFunction:	
m_explicitResult:		false
m_result:				
mscriptToBeRun:	false
No brief
mConsoleRepliesAwaited	0
m_remembered	
wxFile ids - none
No Duktape context
----------------------------------------------------
pJavaScript_pi->mpBin		nullptr

End of dump

Thanks in advance
Jeremy
</div>

antipole2, 2024-04-27:
I've changed this to be an issue - that seems more appropriate given what you are reporting.

You say it crashes OCPN and yet you say the console window stays open.  I do not understand that.

Please update the JavaScript plugin to the current one, which is 3.0.2.  This could be relevant.

The issue is likely because the script uses the degree symbol.  The degree symbol is outside the basic ASCII character set.  We have had terrible problems making the degree symbol usable under Windows. V3.0.2 does does some slight-of-hand to appear to allow a degree symbol.  I did not think it was an issue for Linux - certainly the versions of Linux we have tested on.  Possibly yours is thrown by it.  See the User Guide section 22.

With v3 it should be possible to handle the various characters representing the degree symbol, so I have updated the BuildRaceCourses.js script to use a degree symbol rather than the work-around needed with earlier versions of the plugin - namely  `String.fromCharCode(176)`

So...

1. Update to v3.0.3
2. Take a fresh copy of the script.  With v3 you can now do this by copying the script URL to your clipboard and thenLoad -> URL on clipboard.  The URL is `https://raw.githubusercontent.com/antipole2/JavaScripts-shared/main/BuildRaceCourses/BuildRaceCourses.js`

LordLaingholm, 2024-04-28:
Thanks for that. 

According to the list of plugins in OpenCPN the javascript plugin is 3.0.2 however this is not reflected in the dump. Where do I find version 3.0.3? - I updated the catalog master in plugins but it didn't appear. 

I've done some more testing and pinpointed the issue to the type:"field"... lines in the config dialog. If I comment out these two lines and run the script the dialog box appears. If I leave either of those lines uncommented then it crashes.

I thought it might be caused by the width values so I tried shortening both of these to width:60 - same problem. I tried removing the width:nnn value and it still crashes.

	{type:"field", label:"Pattern", value:"^(Start|Finish|RM)", width:300},
	{type:"hLine"},
	{type:"field", label:"Route name", width:200},

Thoughts?

antipole2, 2024-04-28:
Apologies - my typo - latest is 3.0.2.
In the dump it shows as 3.0 patch 2.

I am wondering about the characters in the value string.
Please run the following one liner:
`print("^(Start|Finish|RM\n");`
You should get

> ^(Start|Finish|RM
> result: undefined

You my wish to create. second console for these ad hoc tests - you can do this in the tools.

antipole2, 2024-04-29:
Here is a variation of the test script.  It displays a message at three points A, B & C.
I am interested how many of those you see before the crash.
NB Point B will be displayed at the same time as the name dialogue.  You would need to dismis Point B before being able to type something in the name field.
````
messageBox("Point A");
onDialogue(process, [{type:"field", label:"name"}]);
messageBox("Point B");

function process(dialogue){
	messageBox("Point C");
	print("Name is ", dialogue[0].value, "\n");
	}
````

LordLaingholm, 2024-04-30:
Point A message box is displayed but after clicking ok it crashes.

where would I find a crash report with traceback?

There are 3 ocpn os processes started.
i@openplotter:~ $ ps -ef|grep opencpn
pi          2816    1106  0 22:30 ?        00:00:00 bwrap --args 38 opencpn.sh
pi          2826    2816  0 22:30 ?        00:00:00 bwrap --args 38 opencpn.sh
pi          2827    2826 35 22:30 ?        00:00:10 opencpn
pi          2956    2306  0 22:30 pts/0    00:00:00 grep --color=auto opencpn

So I started JS and before running the script I started a trace on each process.
pi@openplotter:~ $ nohup strace -p 2816 > 2816.txt &
[1] 2959
pi@openplotter:~ $ nohup: ignoring input and redirecting stderr to stdout

pi@openplotter:~ $ nohup strace -p 2826 > 2826.txt &
[2] 2965
pi@openplotter:~ $ nohup: ignoring input and redirecting stderr to stdout

pi@openplotter:~ $ nohup strace -p 2827 > 2827.txt &
[3] 2973
pi@openplotter:~ $ nohup: ignoring input and redirecting stderr to stdout

Then I ran the script, ocpn crashed and I've attached the trace files.
 
[2827.txt](https://github.com/antipole2/JavaScript_pi/files/15162085/2827.txt)
[2826.txt](https://github.com/antipole2/JavaScript_pi/files/15162086/2826.txt)
[2816.txt](https://github.com/antipole2/JavaScript_pi/files/15162088/2816.txt)

I'm not sure if you can tell anything from these files except it's definately getting killed/aborted.

is there a way to turn on tracing in opcn?

antipole2, 2024-04-30:
Asking @Rasbats for advice here.
Mike... the test script above surely works on the Linux you have tested.  But seems to be crashing OCPN on openplotter.  I am out of my depth here, sailing tomorrow morning from Ulsteinvik. and will have only occasional mobile and no WiFi.
Can you be an angel and either advise or point us to who to consult?  The traces mean nothing to me.

Rasbats, 2024-04-30:
@antipole2 I will have a play. Good sailing.

Rasbats, 2024-05-01:
`I'm running OpenCPN 5.8.4-1 on Linux openplotter 5.15.76-v8+, Raspberry Pi 4`

@Rasbats  can set that up here and see what things look like.

antipole2, 2024-05-01:
If anyone needs to dig deeper, the field is set up in JSdialog.cpp lines 277-340.
It creates wxTextCntl field and at line 319 sets a validator with `textCtrl->SetValidator`

I mention this because I think that is unique to the filed input and other aspects are working without issue, I think.

(Waiting for wind under clear blue sky 😕

Rasbats, 2024-05-01:
@LordLaingholm  Can only find openplotter 3.6.2-stable as the latest version of openplotter. How did you set up the Rpi 4? Did you use a stock image from anywhere? If so a URL would help. Thanks.

LordLaingholm, 2024-05-02:
Hi @Rasbats,
Using the opencpn installer version 3.2.20 I stalled the flatpack version
Flatpak
Installed: 5.8.4-1+1637c28 - stable
Candidate: none

This page in the manual https://openplotter.readthedocs.io/en/latest/opencpn/opencpn_app.html explains the options, I probably installed this version as according to the manual it has the best support for touchscreens.

At the time of writing this manual, OpenCPN Flatpak is the best option for touch screens because it allows right-clicking and two-finger zooming. However OpenCPN Flatpak may not work well in headless environments.

thanks for helping!

LordLaingholm, 2024-05-02:
Hi @rasbats, 

Good question, I got the OP version from running linux command uname -a

pi@openplotter:~ $ uname -a
Linux openplotter 5.15.76-v8+ #1597 SMP PREEMPT Fri Nov 4 12:16:41 GMT 2022 aarch64 GNU/Linux

However it looks like this doesn't reflect the openplotter version as I installed version 3.xx from the links in the online manual - https://openplotter.readthedocs.io/en/3.x.x/getting_started/what_do_you_need.html . In openplotter, settings page it shows settings app version 3.6.6. So I'm probably running 3.6.2 as you mention - how do I find this out? is there a system command I can run?

thanks for your help!

LordLaingholm, 2024-05-02:
btw I'm running Opencpn flatpack, not sure if this makes a difference

Flatpak. Only 64bit. Recommended for: non LTS systems, touchscreens.

	Installed: OpenCPN 5.8.4-1+1637c28 - stable
	Candidate: none

Rasbats, 2024-05-02:
@LordLaingholm  Just about to go out. Probably have to wait until the weekend, when it is raining again!

LordLaingholm, 2024-05-02:
haha all good!

Rasbats, 2024-05-05:
It is raining.

Installed the RPi image frop OpenPlotter. It said 3.6.2 on the website but in settings I seem to have 3.6.6.
The plugin works as expected.
Things to check:

- Set the directory in use with "Tools", to help find a script.
- The waypoint names are case-sensitive! (Tony are you reading!).

antipole2, 2024-05-05:
Limited capacity to attend at mo.  Just dropped anchor late.
Not sure why you say to set directory.  Only needed if running diagnostic suite. 
Can load scripts using Load -> Other.
Important thing to check is that the short test script above works as that showed the issue.

LordLaingholm, 2024-05-06:
Thanks guys for continuing on this, I'll wait for Rasbats response from running the test script, in the meantime I'm wondering if there's any way to get more info out of ocpn like lower level logging, a trace file that might give us an error message or a clue about where exactly this is failing?

LordLaingholm, 2024-05-06:
I ran the test script again and traced the opencpn OS processes, there are 3 processes and 1 of them seems to do all the work, trace file here : [1578.trace.txt](https://github.com/antipole2/JavaScript_pi/files/15219703/1578.trace.txt)

Most of the lines  look like it's just polling but at the end there's this
write(2, "/usr/include/c++/12.2.0/bits/stl"..., 298) = 298
rt_sigprocmask(SIG_UNBLOCK, [ABRT], NULL, 8) = 0
gettid()                                = 2
getpid()                                = 2
tgkill(2, 2, SIGABRT)                   = 0
--- SIGABRT {si_signo=SIGABRT, si_code=SI_TKILL, si_pid=2, si_uid=1000} ---
+++ killed by SIGABRT +++

The /usr/include/c++/12.2.0 directory does not exist, /usr/include/c++/10 does - maybe this is the problem??

I'm not a Linux admin so this is a bit beyond me but I ran an os update thinking this might upgrade c++, the update ran for ages and updated lots of packages and now the kernel is version 6.1.21-v8+ 
pi@openplotter:~ $ uname -r
6.1.21-v8+

Unfortunately the c++ include version has not changed and the problem still exists. 
Rasbats out of interest can you check what directory you have here plz /usr/include/c++/?

Rasbats, 2024-05-06:
test script works for all three point but the name textbox does not accept input.

I am also running 6.1.21-v8+

_can you check what directory you have here plz /usr/include/c++/?_

Under c++ one directory named 10 containing several sub-folders.

LordLaingholm, 2024-05-08:
Thanks Rasbats. I'm going to do a fresh install of OpenPlotter and see if that works.

LordLaingholm, 2024-05-10:
Good news I got the test script working. I did a new install of OpenPlotter using image 2023-08-04-OpenPlotter-v3-Touchscreen-stable-64bit-img.zip. My original install of OP was about 2 years ago using a beta64bit image so maybe something in that was incompatible. 

Next step is to configure OpenCPN, load some s63 charts and add some waypoints then test again.

LordLaingholm, 2024-05-12:
Charts added and imported my routes and waypoints after exporting from opencpn on from backup sd card which is still the old image. 

Fired up JS and ran the build_race_course script which worked perfectly and I added one route.
So far so good however now the script has stopped working, when I run it I get error:
line 4 Error: Console JavaScript - SyntaxError: parse error (line 1, end of input)

These are the first 5 lines
/	This script for the JavaScript plugin provides creation of racing routes by selecting race marks from a list
//	being a subset of OpenCPN waypoints

Position = require("Position");
//	declare enduring variables

I can't see anything obvious in the script, just in case I accidently changed it I downloaded it again from the user guide but same problem, I restarted OpenPCN, bounced Pi - no change.

Any ideas?

Rasbats, 2024-05-12:
Away sailing. Back on Tuesday. Tony may have an idea before then.

antipole2, 2024-05-12:
Do I see a single / on the first line comment?  Comments require //
Copy on file is correct here.

antipole2, 2024-05-14:
> I commented out line 4 which is this.
> Poisition = require("Position");

Please check spelling of first Positionin above

LordLaingholm, 2024-05-14:
what does the line actually mean - can I try coding it a different way?

antipole2, 2024-05-14:
require in JavaScript is a bit like #include.  The module is loaded and the left-hand side becomes whatever was compiled.  In this case, the Position constructor.
Please read the user Guide section 4

LordLaingholm, 2024-05-15:
This has got me stumped. I've removed the script again from the local directory and the compressed file and downloaded it again from the userguide pdf. This is exactly what I did the first time when it worked.

I clear the plugin console and load the script again - 

It errors with
 line 4 Error: Console JavaScript - SyntaxError: parse error (line 1, end of input)

line 4 is:
Position = require("Position");

If I comment out this line the script runs and prompts me for waypoints to build the route, it fails when I select build which is obviously expected 
handleRoute line 165 ReferenceError: identifier 'Position' undefined

I don't understand why I'm getting the initial error when I've run this script successfully the first time and I've downloaded it again so it should be the same. 
Possible causes
- There is a syntax error in the script - hidden characters? caused by the uncompress or ? 
- the script is good then it could be environment? which why i went down the locale rabbit hole - but I thing thats a wrong assumption
- theres something wrong with the require function - there's a few google posts on this -  
- - i uninstalled and reinstalled the JS plugin, same issue
Versions I'm running are:
OpenCPN 5.8.4
JavaScript plugin v3.0.4e

Any other ideas?

antipole2, 2024-05-15:
Probably a character issue.  Please study section 22 of the user guide.
This explains how to find the offending character.
Rasbats and I had to do an enormous amount of work to stop compilation errors under Windows.  Maybe your platform has found another gotcha.
You can use the tool pointed to in that section to see what the cleaning filter is doing with your problematic line.

antipole2, 2024-05-17:
OK so to get you going, I attach a copy of the Position constructor.
Unzip it and store somewhere on your computer.  Then use
Position = require("/where/I/have/put/my/file.js"); (replacing this with your file location)
NB include the .js file extension.
You should be able to use this fix to make progress with your application.
[Position.js.zip](https://github.com/antipole2/JavaScript_pi/files/15345231/Position.js.zip)

Now why is it not working without the above?
`require` without a .js extension causes the plugin to look for the .js file in the library of built-in scripts where the plugin installer should have put them..
Fof example under my MacOS it looks in
`/usr/Tony/Library/Application Support/OpenCPN/Contents/SharedSupport/plugins/JavaScript_pi/data/scripts`
so I could write
````
Position = require("/usr/Tony/Library/Application Support/OpenCPN/Contents/SharedSupport/plugins/JavaScript_pi/data/scripts/Position.js");
````
The plugin asks OCPN for the location of the data folder.  Either that is wrong in your case or the file is not being installed in data/scripts correctly.
Please have a look for this librray and try and establish which of the above it is.
Thanks for your help in pursuing this issue.

LordLaingholm, 2024-05-17:
I tested the Position.js file with the path and same issue. I found where tthe javascripts are stored on the pi here: 
/home/pi/.local/share/opencpn/plugins/JavaScript_pi/data/scripts

There are some other js files in this location like Route.js and Waypoint.js so I substituted those into the command and the script runs so I'm thinking it's something in the Position.js script it doesn't like. 

Would an issue  in the Position.js script generate the syntax error in the console?

Are there any dependencies that the script needs to run. eg It looks like it's looking for latitude and longitude however I'm testing this at home so don't have gps setup - does this matter?

antipole2, 2024-05-17:
Below is a test script that test compiles all the include files.  Please run it.
````
files = ["Position", "Waypoint", "Route", "Track", "canboatAnalyzer", "fibonacci", "File", "NMEA2000", "pgnDescriptors", "shell"];
errors = 0;
for (var f = 0; f < files.length; f++){
	try {
		a = require(files[f]);
		print(files[f], " OK\n");
		}
	catch(err){
		printRed(files[f], " failed ", err.message, "\n");
		errors++
		}
	}
scriptResult(errors + " errors");
````

antipole2, 2024-05-17:
If Position.js is failing to compile on your system, please unzip this copy and run it as a script. 
It should compile and just end with `result: undefined`
If it fails, you should, hopefully, get a line number of the issue or some other clue.
[Position.js.zip](https://github.com/antipole2/JavaScript_pi/files/15355349/Position.js.zip)

LordLaingholm, 2024-05-18:
From the compile script
Position failed Console JavaScript - SyntaxError: parse error (line 1, end of input)
Waypoint OK
Route OK
Track OK
canboatAnalyzer OK
fibonacci OK
File OK
NMEA2000 OK
pgnDescriptors failed Console JavaScript - SyntaxError: parse error (line 1, end of input)
shell OK
result: 2 errors

I opened the script in Visual Studio and noticed these weird characters highlighted, the "\<degree>" with a message that these are not ascii characters, except of the code here

("00" + parseInt(latAbs)).slice(-2) + "\º " +
            ("00" + (latAbs % 1 * 60).toFixed(3)).slice(-6) + "'" +
            ((this.latitude < 0) ? "S " : "N ") +
            ("00" + parseInt(lonAbs)).slice(-3) + "\º " +

However when I ran the script there were no errors just this message
result: undefined

I ran the build course script again and it failed with the same error.

As a test I replaced the degree symbols with a '*', there are 4 in the Position script and ran build courses and the first popup appeared - no error!

So this brings me back to locale and a character set issue which is currently set to this:
pi@TimeOut-pi:~ $ locale
LANG=en_NZ.UTF-8
LANGUAGE=en_NZ.UTF-8

I changed it to 
pi@TimeOut-pi:~ $ locale
LANG=en_US.UTF-8
LANGUAGE=en_US.UTF-8

and reversed the changes to the script and bingo! it's working now in that the first popup appears without error. I'll test it fully tomorrow as it's getting late here and hopefully this is all fixed. 
I ran the compile script again and that looks good now
result: exited
Position OK
Waypoint OK
Route OK
Track OK
canboatAnalyzer OK
fibonacci OK
File OK
NMEA2000 OK
pgnDescriptors OK
shell OK
result: 0 errors

Rasbats, 2024-05-18:
Brilliant Jeremy. Tony should be able to fix this. We had many similar problems elsewhere with the degree symbol.

antipole2, 2024-05-18:
Yes, well done!
As Mike says, the degree symbol has been a real pain.  In fact, for Windows we have to substitute it with another character and the switch back for all outputs!
Linux handles it OK but, clearly, NZ codes are different.

We have just arrived in Bergen and are heading for the bunk.  But please do the following test so I can consider how to,proceed:
With the en_NZ.UTF-8 selected, open the Diagnostics tab,of tne tools.
The field should contain ‟Fancy quotes” degrees°º⁰ primes ‘’‛’′´`
Note all the different representations of degree in it,
Clean and let me have the output.
[Should the three degree symbols not be shown after the word degree please type some in and let me know what you have done.  In this case a screen shot would be helpful.]

LordLaingholm, 2024-05-19:
Here's the output from diagnostics clean, I added the degree symbol from the postion script, it looks like it's the same as he 2nd symbol?
[0000]‟ [01]F [02]a [03]n [04]c [05]y [06]  [07]q [08]u [09]o 
[10]t [11]e [12]s [0013]” [14]  [15]d [16]e [17]g [18]r [19]e 
[20]e [21]s [22]° [23]º [0024]⁰ [25]º [26]  [27]p [28]r [29]i 
[30]m [31]e [32]s [33]  [0034]‘ [0035]’ [0036]‛ [0037]’ [0038]′ [39]´ 
[40]` 

Raw
‟Fancy quotes” degrees°º⁰º primes ‘’‛’′´`
[00]201F [01]46 [02]61 [03]6E [04]63 [05]79 [06]20 [07]71 [08]75 [09]6F 
[10]74 [11]65 [12]73 [13]201D [14]20 [15]64 [16]65 [17]67 [18]72 [19]65 
[20]65 [21]73 [22]B0 [23]BA [24]2070 [25]BA [26]20 [27]70 [28]72 [29]69 
[30]6D [31]65 [32]73 [33]20 [34]2018 [35]2019 [36]201B [37]2019 [38]2032 [39]B4 
[40]60 

Cleaned
"Fancy quotes” degrees°º⁰º primes ‘’'’′´'
[00]22 [01]46 [02]61 [03]6E [04]63 [05]79 [06]20 [07]71 [08]75 [09]6F 
[10]74 [11]65 [12]73 [13]201D [14]20 [15]64 [16]65 [17]67 [18]72 [19]65 
[20]65 [21]73 [22]B0 [23]BA [24]2070 [25]BA [26]20 [27]70 [28]72 [29]69 
[30]6D [31]65 [32]73 [33]20 [34]2018 [35]2019 [36]27 [37]2019 [38]2032 [39]B4 
[40]27 

btw the tools headings are hard to read as they're white against an offl white background

antipole2, 2024-05-19:
Very interesting.  I need to explain:
The degree symbol is variously represented as 
˚     degree                    \xB0
⁰    masculine ordinal  \xBA
º    superscript zero    \x2070

Text editors often use different characters without it being obvious to the user.
The plugin cleans the input by translating masculine ordinal and superscript 0 to degree \xBA.

There is a similar problem with various primes and quotes.

When I run the clean string test I get 

Cleaned
“Fancy quotes” degrees°°° primes ‘’’’’’’
[00]22 [01]46 [02]61 [03]6E [04]63 [05]79 [06]20 [07]71 [08]75 [09]6F 
[10]74 [11]65 [12]73 [13]22 [14]20 [15]64 [16]65 [17]67 [18]72 [19]65 
[20]65 [21]73 [22]B0 [23]B0 [24]B0 [25]20 [26]70 [27]72 [28]69 [29]6D 
[30]65 [31]73 [32]20 [33]27 [34]27 [35]27 [36]27 [37]27 [38]27 [39]27 

All the degree variants have been translated to \xB0 and primes to \x27

Your degrees are not being translated and only a few of your primes.
I have no idea why.  It is possible that NZ is using different codes for some - perhaps making way for Mōari characters?
I am on a mobile signal and small screen and not in a position to get to the bottom of this right now.
I cannot understand why \xBA and \x2070 are not being translated to \xB0.  The code has no Linux dependency and works with UK/USA character set.

However, I can help you get going.  In the recent versions of Position.js I reverted to using \º which is actually the masculine ordinal and is failing to be translated with the NZ character set.
So... in Position.js, replace all occurences that look like  \º with \b0

Hopefully you cqn get going with that.

If you have the inclination to dig deeper, the code for the cleanString function is in [this file](https://github.com/antipole2/JavaScript_pi/blob/master/src/functions.cpp) lines 62-112

Sorry I cannot do more t the moment

antipole2, 2024-05-20:
Whops _ replace with \xb0

LordLaingholm, 2024-05-22:
thanks for that, I haven't had a chance to test it yet as I left the pi on my boat which is 30 mins away and I'm working full time, will get down there this weekend and test it

LordLaingholm, 2024-05-24:
Bingo! it's working now with my locale set to nz and I'm building routes so far so good. 

Thank you so much for your help, hopefully this issue can be closed now :-)

LordLaingholm, 2024-05-25:
I can't see how to close this issue?

antipole2, 2024-05-25:
You add a comment like this but instead of clicking on Comment you pull down on ‘Close with comment’.  I will close it now.

closed 2024-05-25 by antipole2