antipole2/JavaScript_pi #102: Consle not seen after disabling and re-enabling plugin

antipole2, 2024-06-02:
@torrmundi reports
`If I disable/reenable Javascript PI, clicking on the {JS} icon no longer brings up my console. Nothing happens, except the icon goes grey (on) or black (off) on succcessive clicks. I have to stop/restart OpenCPN to regain a working console. Then it behaves as before: 1st try no responses, 2nd and onward gets responses. regain a working console. Then it behaves as before: 1st try no responses, 2nd and onward gets responses.`

Looks like you have uncovered an issue whereby console is positioned out of sight.  May arise because of your screen resolution.  To help me investigate, please do the following:

1. With plug-in behaving correctly, take dump, capturing all data.
2. Disable plugin (by unticking in the list of plugins)
3. Capture your OCPN config file OpenCPN.ini
4. Re-enable the plug-in and open the tools via the Preferences button in the plugin entry
5. Find Diagnostics tab and take another dump, capturing all data.
6. Go to the Consoles tab in tool and pull down on the ‘Change name of console, list.  Can you see the names of your consoles in this list(even though you cannot see the consoles)?

Suggest you zip the three files to avoid long reply in this thread.
Thanks for your help with this.

antipole2, 2024-06-02:
Something else to check/try…. In Consoles tab of Tools bottom left is an option `Float consoles on top`.  Is that unselected?  If you turn it on, does that change the behaviour - I.e. do you then see the console after re-enabling the plugin?

torrmundi, 2024-06-02:
New Issue: The Tools Consoles tab does have that Float... button, but it's hidden until I drag the dialog bottom downwards, exposing the two checkboxes.  My screen is 1920x1200 (highest available) and my Scale is 125%.

I didn't have the float option enabled. But enabling it doesn't change behavior. It's also the same if the console is parked, or open when I disable the PI.

torrmundi, 2024-06-02:
Something different this time. When I re-enabled, my console suddenly reappeared.
Files are attached.
Step 6: Yes, the name (WhoGoesThere) is in the dropdown.
[WhoGoesThere 1.zip](https://github.com/user-attachments/files/15527224/WhoGoesThere.1.zip)
I'll try this again. I don't think I did anything different, except Step 4.

torrmundi, 2024-06-02:
Weird. Now the console pops up every time I re-enable (or the parked console pops up). But, the Preferences button no longer brings up a dialog. I restarted OCPN. Opened the console, ran the script (ok), Opened the Preferences (ok). Closed Preferences.
I disabled the PI. Re-enabled, opened Preferences (ok).  Continuing, got the dump. Console name is in the dropdown.
[WhoGoesThere2.zip](https://github.com/user-attachments/files/15527311/WhoGoesThere2.zip)
FYI, When things go well, Alt-Tab shows windows for the Tools, the Console, and OpenCPN. When things are not showing up, only OpenCPN is available via Alt-Tab. That makes me think the window isn't hidden - rather, it doesn't exist.

antipole2, 2024-06-03:
> New Issue: The Tools Consoles tab does have that Float... button, but it's hidden until I drag the dialog bottom downwards, exposing the two checkboxes. My screen is 1920x1200 (highest available) [...]

[Separate issue created for this one](https://github.com/antipole2/JavaScript_pi/issues/103).

antipole2, 2024-06-03:
@torrmundi  Please confirm whether when you report Preferences button not working, is this in a console (Tools) or in the Plugins list 'Preferences' button?
The dumps you sent show the console being restored in the same position as was before disabling.

If the Preferences button in the plugins entry is not working, it suggests the plugin was not being re-enabl;ed correctly - hence the consoles not appearing.  Please send the relevant bit of the log when this happens.  It should be logging the initialisation process.

I am on board and mostly working on a mobile signal and small screen, so bit limited on how deep I can go.

torrmundi, 2024-06-03:
@antipole2 I believe I got the dump you want, with pref not working as accessed from the PI list. Included a screen cap.
[Javascript PI Prefs.zip](https://github.com/user-attachments/files/15534292/Javascript.PI.Prefs.zip)

antipole2, 2024-06-03:
In the screenshot you sent, the JavaScript plugin is not enabled.  The Enabled box is not ticked.  Hence Preferences is greyed out.

On my system if I untick this, the JS icon in the left hand bar does not go away until I dismiss the options panel.

When I tick the Enable box, the consoles seem to appear but the JS icon does not appear until I dismiss the options panel.

The options panel is modal.  It blocks many other processes until it is dismissed.  Are you dismissing it after enabling before trying to use the plugin?

torrmundi, 2024-06-03:
I'm not always dismissing the PI List.
Try this:
When I click on the Javascript PI in the list, it opens up the list to show
the options like Preferences, but it also unchecks the enabled button.
Open PI in the list, disable JPI.
OK.
{JS} is in the console still, toggles black/gray, does nothing when clicked.
Open PI List, enable JPI.
Parked WhoGoesThere appears.
OK
Open PI List, open JPI entry, disable.
OK
{JS} still in console, still does nothing.
Open PI List, open JPI entry, enable.
Preferences button goes gray to black.
Click on Preferences. Nothing.
OK
Open PI List, open JPI entry
Click on Preferences. Preferences dialog opens.

antipole2, 2024-06-03:
The {JS} black/grey button:
- when toggled to black it hides all consoles
- when toggled to grey it shows all consoles (may be parked).
- by default, when the plugin starts up, the button is black with **all consoles hidden**.  You need to click to make it grey to see the consoles.
- The `Remember toggle status` option (next to the `Float consoles on top`) overrides that such the toggle status is preserved across restarts.
- On my Mac the toggle status does not get changed until I dismiss the options window.  I cannot click on the toggle until I do.  If you can click on it, it may respond as if the colour had changed, even though the has not yet changed.

I suspect most of the oddities you list are because you are not clicking Apply or closing the panel thus allowing OCPN to complete the changes you have made, which includes refreshing the window so it shows what it should in the right colour/shade or hides what it should not show.

antipole2, 2024-11-28:
In plugin v3.1, the default behaviour for a first-time user preserves the Toolbar toggle status of the plugin.
The user can change this behaviour in the tools if desired.  Having done so, they should be aware of the effect.