antipole2/JavaScript_pi #103: Tools > Console tab not showing the Float option etc, on higher resolution Windows displays unless window extended downwards

antipole2, 2024-06-03:
[torrmundi](https://github.com/torrmundi) reports:

>  The Tools Consoles tab does have that Float... button, but it's hidden until I drag the dialog bottom downwards, exposing the two checkboxes. My screen is 1920x1200 (highest available) and my Scale is 125%.

I had to do an extraordinary amount of work to handle higher resolution displays. It seems I have not covered this bit.  Maybe I added the Float option after I had fixed the screen resolution issue.

MacOS handles screen resolution internally and no action is needed.  I had to write a simulator to develop the fixes for Windows, which have to be applied to each window layout.

This is a job for next winter, when I am not afloat.

antipole2, 2024-06-03:
@torrmundi Please let me have a screenshot of

1. the tool Consoles tab with the Float… button not showing.  Just as it is when you first open the tab
2. The window extended down enough to show the two options not previously visible

thanks

torrmundi, 2024-07-03:
Ok, finally revisiting this old issue.  Here's the screen captures.
[image]
[image]

torrmundi, 2024-07-03:
I'm going to risk "teaching a dog to bark" here. But this is a good discussion of scaling in Javascript for various browsers.
https://stackoverflow.com/questions/2242086/how-to-detect-the-screen-resolution-with-javascript

antipole2, 2024-07-03:
> I'm going to risk "teaching a dog to bark" here. But this is a good discussion of scaling in Javascript for various browsers. https://stackoverflow.com/questions/2242086/how-to-detect-the-screen-resolution-with-javascript

Always happy to learn some new tricks.  However, the plugin's environment is very different from a brower's.  There is no `window` structure on which those solutions depend.
The plugin does know the resolution, but I have not made it available to scripts.  You can inspect it via the tools->dump.  The third line is `JavaScript tools window DPI scaling factor 1.000000` for me on a Mac.

antipole2, 2024-11-30:
Alpha build of v3.1 of plugin has re-worked Tools panes.  Hopefully display on Windows hi-res displays OK?
@torrmundi Are you in a position to give it a try?
You can download build [c2f27d0](https://cloudsmith.io/~opencpn/repos/javascript-alpha/packages/) from my alpha repo.
Attached. are  screenshots of how it looks on MacOS.
Really grateful if you could check this out.
[Tools_images.zip](https://github.com/user-attachments/files/17965547/Tools_images.zip)

torrmundi, 2024-11-30:
Tony,

Unfortunately, I'm not on my boat now. I'll be happy to test when I get
back, or possibly sooner if I can use the VDR Replay feature.
Please remind me of the proper package to download for a Windows 11
platform - is it the "javascript-3.1-msvc-wx32-10-tarball"?

Regards,
torrmundi

antipole2, 2024-11-30:
@torrmundi
If you have the computer with hi-res screen at home, you don't need to be on the boat.  Sufficient to install and enable the plugin and open the tools.
I assume the msvc tarball is what you need to download and import.
thanks

torrmundi, 2024-12-01:
Tony,
I upgraded 2 PCs (one low res, I see Float... without adjustment of the
window, the other the 4k screen that had the issue) to O 10.5.2.
Then uninstalled Javascript_PI and imported Alpha 3.10 on each.
I can see the Float... checkbox on both without changing the Consoles
window size.
Great! Anything else to check?

Thanks,
John

antipole2, 2024-12-01:
Thanks.  Can you confirm the other tabs in the tools are good too?  The screen shots I included above show what each should have available.

torrmundi, 2024-12-01:
Tony,

Here are my screenshots.  You'll see that Windows' Display Settings>Scaling
makes a difference.
I normally use the 1920x1200 resolution (full resolution of this laptop)
with 125% scaling (for my old eyes).

[image: image.png]

antipole2, 2024-12-01:
Sorry - cannot see images -only screenshot of an email message.

From your message I gather something not yet right.
Please advise scaling factor seen by plugin in each case.
Tools -> Diagnostics -> Dump
The 3rd line of the dump reveals the scaling factor. e.g. `JavaScript tools window DPI scaling factor 1.000000`

torrmundi, 2024-12-01:
Tony,
You should find the attachments to my last email, yes? Those are zip files
of images, similar to what you sent to me.
The only embedded image was the Windows display settings, which I explained
anyway.
Javascript3.10_4kres_100scale.zip = 100% scaling factor
Javascript3.10_4kres_125scale.zip = 125% scaling factor
https://www.windowscentral.com/how-set-custom-display-scaling-setting-windows-10

There seem to be issues with 100% scale for the Diagnostics, and NMEA tabs.

torrmundi, 2024-12-01:
Tony,

Sorry about the attachments, looks like replying via gmail thru Github doesn't totally work.
[Javascript3.10_4kres_100scale.zip](https://github.com/user-attachments/files/17969690/Javascript3.10_4kres_100scale.zip)
[Javascript3.10_4kres_125scale.zip](https://github.com/user-attachments/files/17969691/Javascript3.10_4kres_125scale.zip)
JavaScript tools window DPI scaling factor 1.000000

John

antipole2, 2024-12-01:
Got the screenshots.
The 1.00 ones look OK, although it looks like Windows is screwing up the characters in the sample dump string still.  That's another matter.
The 125 scale is not right but I need to know what the `JavaScript tools window DPI scaling factor 1.000000` value is for that case.  Last time we looked at this it showed the scale, i.e. 1.50000.

torrmundi, 2024-12-01:
With Windows scale = 125%, 
JavaScript tools window DPI scaling factor 1.250000

Is there a missing font pack I should have?

John

antipole2, 2024-12-01:
I now have what I need to look at this further. Thanks.
Will get back to you.

antipole2, 2024-12-02:
The tabs for NMEA and Diagnostics were too tight.  They both have buttons as last thing.  I am now scaling them and it looks OK when I simulate a hi-res display.
Please give it a go.  Build [10f007d](https://cloudsmith.io/~opencpn/repos/javascript-alpha/packages/).

antipole2, 2024-12-09:
@torrmundi As above, can you please confirm the buttons on the NMEA and Diagnostics now OK on hi-res display?

torrmundi, 2024-12-09:
The Diagnostics and NMEA lower edgses are cut off:
[image]
[image]
Using javascript-3.1.0-alpha+425.d98620e_msvc-wx32-10-x86.tar.gz
Same with javascript-3.1.0-alpha+424.10f007d_msvc-wx32-10-x86.tar.gz

torrmundi, 2024-12-11:
at any scaling factor from 100% to 150%, all the tabs look good. But at 175%, a problem shows up on Consoles tab:
[image]
At 200%, it actually a bit better, but is still a bit cut off

antipole2, 2024-12-12:
@torrmundi The issue sems to be that when I read the size of the pane so I can scale it for hi-res displays, if there are trailing buttons, the reported size does not accurately relefct the size.  Some very obscure wxWidgest issue.
I fudged this for NMEA and Dignostics by adding an empty sizer after the trailing buttons so they are no longer trailing.
I have now added the same for the Consoles tab.  Build ee0eb61

torrmundi, 2024-12-13:
OK, looks good at all scaling factors and all tabs!  
Thanks