antipole2/JavaScript_pi #110: Windows mangling fancy characters again

antipole2, 2024-12-02:
I thought we had fixed this tiresome Windows issue - grrr!

Screenshots from @torrmundi show the problem again.

Here is the test string from the diagnostics tab as seen on a Mac:
```
[0000]‟ [01]F [02]a [03]n [04]c [05]y [06]  [07]q [08]u [09]o 
[10]t [11]e [12]s [0013]” [14]  [15]d [16]e [17]g [18]r [19]e 
[20]e [21]s [22]° [23]º [0024]⁰ [25]  [26]p [27]r [28]i [29]m 
[30]e [31]s [32]  [0033]‘ [0034]’ [0035]‛ [0036]’ [0037]′ [38]´ [39]` 

Raw
‟Fancy quotes” degrees°º⁰ primes ‘’‛’′´`
[00]201F [01]46 [02]61 [03]6E [04]63 [05]79 [06]20 [07]71 [08]75 [09]6F 
[10]74 [11]65 [12]73 [13]201D [14]20 [15]64 [16]65 [17]67 [18]72 [19]65 
[20]65 [21]73 [22]B0 [23]BA [24]2070 [25]20 [26]70 [27]72 [28]69 [29]6D 
[30]65 [31]73 [32]20 [33]2018 [34]2019 [35]201B [36]2019 [37]2032 [38]B4 [39]60 

Cleaned
“Fancy quotes” degrees°°° primes ‘’’’’’’
[00]22 [01]46 [02]61 [03]6E [04]63 [05]79 [06]20 [07]71 [08]75 [09]6F 
[10]74 [11]65 [12]73 [13]22 [14]20 [15]64 [16]65 [17]67 [18]72 [19]65 
[20]65 [21]73 [22]B0 [23]B0 [24]B0 [25]20 [26]70 [27]72 [28]69 [29]6D 
[30]65 [31]73 [32]20 [33]27 [34]27 [35]27 [36]27 [37]27 [38]27 [39]27 
```
and here is what Windows has done with it:
[image]

The filtering is in the function `JScleanString` located [here](https://github.com/antipole2/JavaScript_pi/blob/v3.1/src/functions.cpp) in the putative v3.1 of the plugin.  (I have not changed anything in this function since v3.0)

@Rasbats have you any idea what is going on, please?

antipole2, 2024-12-03:
The test string was being set up in wxFormBuilder and the fancy characters compiled.  I think the Windows JavaScript engine chokes on what the compiler has put in there.
So now, when the Diagnostics tab is selected, I am explicitly loading the string using escape sequences.  I hope this will improve the situation.
@torrmundi - please try this.  Please supply a screenshot of the diagnostics tab and also of the text window generated when the Clean button is clicked.
Use build d98620e or later from the [alpha repository](https://cloudsmith.io/~opencpn/repos/javascript-alpha/packages/).

torrmundi, 2024-12-03:
half-way there....
[image]

FWIW, I tried to recreate the test strings.
in Hex mode on Notepad++
[image]

in MS Word, same string:
[image]

John

antipole2, 2024-12-03:
I have been through this very carefully.
The [] are correct - bell is used internally to represent the degree as Windows cannot process the degree. The bell is displayed as [] so that is correct under the circumstances.
Some of the various ‘’‛’′´` have been fixed.  The ? are characters with no representation in Windows which has substituted a real ? for the hex code by the time we get them.  There is nothing more that can be done  - apart from moving on from Windows ;-|)

I propose to mark this 'Live with it'
Thanks for the help.