antipole2/JavaScript_pi #119: Enhancement: SHOW or HIDE NMEA Debug and AIS Target List Windows

plfinch, 2025-04-05:
Create ability to show or hide/close the OpenCPN NMEA Debug and AIS Target List windows from the javascript plugin. There may also be other OpenCPN windows I am not familiar with yet or in the future that this would be useful for so I propose a generic solution rather than new functions for each such window.

OCPNwindowShow(windowName)
OCPNwindowHide(windowName)

Where windowName is the string "NMEA Debug" or "AIS Target List".

I want to be able to programmatically from javascript open and close these windows as needed.

antipole2, 2025-04-05:
Please explain more about why you need this or what you are wanting to achieve.

plfinch, 2025-04-05:
I have a land-based AIS monitoring and reporting station and use OpenCPN for viewing the nearby vessels reporting AIS data.

1. On boot or restart of OpenCPN I want to automatically display the NMEA Debug and AIS Targets windows to verify proper operation and then close the windows after 5 minutes.

2. Anytime AIS data is not received for 2 minutes I want the NMEA Debug window to open until resolved.

3. Anytime a new vessel is acquired I want the AIS Targets window to display for 20 seconds.

I already have the javascript plugin installed and use it for various housekeeping tasks as well as highlighting new vessels.

antipole2, 2025-04-06:
This would be better handled as a discussion for now.  I have[ transferred it here](https://github.com/antipole2/JavaScript_pi/discussions/120)

Closing as an issue for now.

closed 2025-04-06 by antipole2